j-kaltes/Juggluco #132: Juggluco WearOS 7.6.1. build, Juggluco mobile stopped show the watch in Watch > WearOS > Config

vstoianov, 2024-03-11:
I have successfully build Juggluco with build variant wearLibre3Release. I can settup Mirror on the watch and on the Smartphone to exchange data, but In the Android application (Juggluco mobile) I cannot see my watch listed in the WearOs config. If I download a build from your Google Drive the watch is listed and I can use direct connection between the watch and the Libre 2 sensor. I follow all your instructions how to build the application, but something is missing. I checked and I saw there differences between the /lib/armebi-v7a/libnative.so and /lib/armebi-v7a/libg.so from my build and your last apk. How this could be resolved?

j-kaltes, 2024-03-11:
If you have the mobile phone app from Google Drive and compiled the WearOS app yourself, the APKs are signed with a different key.

vstoianov, 2024-03-11:
Hmm, this is correct. I will try to build the mobile version and I will check what will happened.

sidouzi123, 2024-03-15:
Hello,

I noticed that you have successfully built the Wear OS app. I have successfully built the Android app, but I can't find the method to build the Wear OS app. Could you please teach me how to do it?

Thank you!

vstoianov, 2024-03-18:
Hi, sorry for the late reply.
I have just cloned the repository and just used Android Studio with
appropriate Build Type release, Flavors: wearOs > wear. Of course, you must
build the project and "Build > Generate Signed Bundle/APK". Select "APK",
then choose your keys to sign the APK and then select the Build Variant. In
my case I did it successfully using wearLibreOldRelese(supports
Libre1/Libre2 sensors) and using wearLibre3Release(supports
Libre1/Libre2/Libre3 sensors).

If you have any questions we will be in touch.
Reagrds,

-- 

Vasil Stoianov
Technical Manager
CM2W JSC
www.cm2w.net

IMPORTANT NOTICE: This e-mail or fax, and the attached documents if any,
may contain confidential information. The information is intended only for
use by the owner(s) of the listed e-mail address(es) or fax number(s). The
present email shall not constitute conclusion of any agreement between the
parties. In order for the agreements to be binding, they must be concluded
in written form, signed by the intended parties. No obligations can arise
for CM2W following the discussions through email. CM2W reserves all
Intellectual property and Industrial property rights that may arise out of
this email, including but not limited to rights over devices, software,
projects, schemes, calculations, operation method or model, server (mobile)
software, where no action shall be understood as granting such, except
after explicitly stated and agreed by the parties by a signed written
document. No legal claims can be deduced from this e-mail.

If you are not the intended recipient, you are hereby notified that any
disclosure, copying, distribution, or the taking of any action in reliance
on, or regarding the contents of this e-mailed information is strictly
prohibited. If you have received this e-mail or fax in error, please notify
CM2W immediately.