j-kaltes/Juggluco #133: Unable to Build Wear OS App

sidouzi123, 2024-03-15:
Hello,

I have successfully compiled the Android app for Juggluco, but I am encountering issues when trying to build the Wear OS app. I am unsure how to properly proceed with the build and would appreciate some guidance.

Thank you.

sidouzi123, 2024-03-15:
To accurately say, I can't find the source code for the WearOS app.

j-kaltes, 2024-03-15:
The source is the same. It is a different build variant: _wear_ instead of _mobile_.

sidouzi123, 2024-03-15:
> The source is the same. It is a different build variant: _wear_ instead of _mobile_.

I am not very familiar with the use of Android Studio. I found it in Build Variants. very very thankyou. May I ask what is the difference between wearlibre3release and wearlibre3release 2?

j-kaltes, 2024-03-15:
You can find that in Common/build.gradle:
release2 has ".debug" added to the applicationID. It is just the release version with a different applicationID.

sidouzi123, 2024-03-15:
> You can find that in Common/build.gradle:
> release2 has ".debug" added to the applicationID. It is just the release version with a different applicationID.

First of all, thank you for your answer. 😊 I have solved the above problem. I also need to ask a question about translation. I found translation guidelines in another project of yours. I am Chinese and use Chinese. I have already translated it, but how do I add the Chinese option in the language options? 🤔

closed 2024-03-15 by j-kaltes