j-kaltes/Juggluco #134: Add Chinese language option

j-kaltes, 2024-03-15:
https://github.com/j-kaltes/Juggluco/issues/133#issuecomment-1999823284
`I also need to ask a question about translation. I found translation guidelines in another project of yours. I am Chinese and use Chinese. I have already translated it, but how do I add the Chinese option in the language options? 🤔`

At multiple places Chinese has to be added. I can easily do that. I already made some modifications for the Chinese font that also has to be turned on.
You only have to somehow upload the translation https://github.com/j-kaltes/Jugglucohelp, https://raw.githubusercontent.com/j-kaltes/Juggluco/primary/Common/src/main/res/values/strings.xml and https://raw.githubusercontent.com/j-kaltes/Juggluco/primary/Common/src/main/cpp/curve/jugglucotext.cpp

sidouzi123, 2024-03-15:
> https://github.com/j-kaltes/Juggluco/issues/133#issuecomment-1999823284
> `I also need to ask a question about translation. I found translation guidelines in another project of yours. I am Chinese and [...]

Hello!
I have already translated these two files, but I want to add the Chinese option to the language options. It seems that it's not just about translating these two files to achieve this.🥺

sidouzi123, 2024-03-15:
> You didn't translate the help files at https://github.com/j-kaltes/Jugglucohelp ?

sorry.
It might be a language barrier issue, which makes it difficult for me to understand. Do I need to create a pull request to merge the translated files into the "zh" folder in the "jugglucohelp" project?😭

j-kaltes, 2024-03-15:
You already have a translation of https://github.com/j-kaltes/Jugglucohelp ? If as I said you want you name there, you need to do that. But you can also first try it out at home. I will give you the changes you have to make, but I don't know whether I missed something or made a mistake:
```diff
diff -r Juggluco/Common/src/main/java/tk/glucodata/settings/Settings.java /home/jka/src/android/Glucodata/Common/src/main/cpp/../../../src/main/java/tk/glucodata/settings/Settings.java
475c475
< static private final List<String> supportedlanguages= IWRU?Arrays.asList("Language","be","de","en","es","fr","it","iw","nl","pl","pt","ru","uk"):(SPANISH?Arrays.asList("Language","be","de","en","es","fr","it","nl","pl","pt","uk"):Arrays.asList("Language","be","de","en","fr","it","nl","pl","pt","uk"));
---
> static private final List<String> supportedlanguages= IWRU?Arrays.asList("Language","be","de","en","es","fr","it","iw","nl","pl","pt","ru","uk","zh"):(SPANISH?Arrays.asList("Language","be","de","en","es","fr","it","nl","pl","pt","uk","zh"):Arrays.asList("Language","be","de","en","fr","it","nl","pl","pt","uk","zh"));
diff -r Juggluco/Common/src/main/res/xml/locales_config.xml /home/jka/src/android/Glucodata/Common/src/main/cpp/../../../src/main/res/xml/locales_config.xml
11a12
>    <locale android:name="zh"/>

--- curve.cpp	2024-03-15 16:00:49.935877208 +0100
+++ /u/home/jka/src/Juggluco/Common/src/main/cpp/curve/curve.cpp	2024-03-15 16:15:49.219220429 +0100
@@ -2676,13 +2676,14 @@
 				}
 			break;
 			#endif
+			/*
 		case mklanguagenum("ZH"):
 		case mklanguagenum("zh"):
 			setusezh();
 			if(chfontset!=CHINESE) {
 				initfont();
 				}
-			return; 
+			return; */
 		default: setuseeng();
 		};
 	if(chfontset!=REST) {
```

sidouzi123, 2024-03-15:
> You already have a translation of https://github.com/j-kaltes/Jugglucohelp ? If as I said you want you name there, you need to do that. But you can also first try it out at home. I will give you the [...]

Okay, after I test and proofread the text, I will send it to you by email and create a pull request for the "jugglucohelp" project.😉😉😉😉😉😉

This issue can be closed. Wishing you a pleasant work and life.Thank you very much for answering my question.

j-kaltes, 2024-03-15:
I don't know if you found it out yourself already, but for Juggluco to make use of these help files, you also need to compile html2xml. Copy pl/Makefile to zh/Makefile and replace pl with zh and change the paths in Makefile to the locations on your computer. Then you have to call `make install` at the command line.

sidouzi123, 2024-04-16:
This issue has been resolved and can be closed.

closed 2024-05-23 by j-kaltes