j-kaltes/Juggluco #136: CORS in web server?

jmandel, 2024-03-17:
Thanks for this incredible app!

Has there been consideration of adding an option for CORS support in the built-in web server? I'd like to explore building a static web app that can connect to my local juggluco over http and (with an appropriate API key) pull data from the app. But this would require CORS headers in the juggluco web server response.

Details at https://cors-test.codehappy.dev/?url=http%3A%2F%2F127.0.0.1%3A17580%2Fx%2Fstream%3Ftoken%3Dsecret&origin=https%3A%2F%2Fcors-test.codehappy.dev%2F&method=get

I'd be happy to work on a contribution if that's helpful.

j-kaltes, 2024-03-18:
I made some changes, and according to https://www.test-cors.org/ it works with CORS.
Here is an URL to the new version:
https://drive.google.com/file/d/12b63DcsUbcxFZJhR2YsUOBJEhBq9dF0D/view

jmandel, 2024-03-18:
Just for ease of testing, here's an app that shows a table or chart of past 24h strema:

Source: https://glitch.com/edit/#!/gluco?path=index.html
Hosted: https://gluco.glitch.me

j-kaltes, 2024-03-18:
I don’t see how this CORS restriction on what browser do can contribute to privacy or security. If you fill in the api-secret they can do with it whatever they want. These CORS nonsense will not help you. You have to trust the maker of website.

jmandel, 2024-03-18:
The issue is ensuring that people who turn on the web server but do not create a secret (or create an insecure secret) are protected from web sites scraping their data.

j-kaltes, 2024-03-19:
Yes, now it is the case that when you specify _local only_ and no api_secret, an arbitrary site you visit can get access to the data. But restricting this to particular websites would be very difficult to configure for the user: They need to enter the urls of the websites from which it is allowed to access the data.

jmandel, 2024-03-20:
I agree, that's a reasonable approach and consistent with my "at least one of" suggest at https://github.com/j-kaltes/Juggluco/issues/136#issuecomment-2004329732

Thanks for the ultra quick prototype and the good discussion here!

jmandel, 2024-03-20:
Reading more closely, I do think the API key should be required i*rrespective of the local-only flag.* People have some reason to understand that the networking and firewalling around their device can prevent it from being accessible by arbitrary outside parties..  but CORS totally breaks that assumption because it makes their server accessible from any website

j-kaltes, 2024-03-20:
I have uploaded a new version to https://www.juggluco.nl/Juggluco/download.html that requires a api_secret larger than 7.

closed 2024-09-22 by jmandel