j-kaltes/Juggluco #148: Are libre3 device IDs preserved when uploading?

jmandel, 2024-04-25:
https://github.com/j-kaltes/Juggluco/blob/e4f2c50f722c8c3a49a17ab0409b1a55fba25a5e/Common/src/mobile/java/tk/glucodata/Libreview.java#L104

This code in some circumstances mints a new ID, and I'm wondering if this could be the reason the IDs I see in Libreview CSV download don't match my device serial number. I don't understand the system well, so maybe these aren't even expected to match?

j-kaltes, 2024-04-25:
These device IDs are always arbitrary numbers used to differentiate between different uploaders. The first time they are generated with the sited procedure saved there. At later times the same number is retrieved. It is useful for knowing who did uploading and you can selectively delete the values uploaded by a particular uploader:
[image]

jmandel, 2024-04-25:
Ah. Thanks -- I was expecting my device serial number (the one printed on the device and which appears in the juggluco main UI) to appear somewhere in the CSV export from Abbott, but maybe just don't service that information.

closed 2024-04-25 by jmandel

j-kaltes, 2024-04-25:
In the main user UI of Juggluco the sensor serial number is shown. 
The device ID is specific for a particular app on a particular phone. It stays the same when you switch to a new sensor.