j-kaltes/Juggluco #151: Scan Error when trying to first connect to Freestyle Libre 3

Schabolon, 2024-04-29:
Hi,

I just got the Freestyle Libre 3 sensor and wanted to connect to Juggluco.
All permissions were granted and I manually set a Account-ID in the Libreview-Setting.
When I hold my phone close to the Libre 3 sensor, I get the following message: "FreeStyle Libre 3, Scan error, Try again"
(Similar to the last posts in #13 from @Venipa)

Phone Model: Samsung Galaxy S10+
Custom ROM: LineageOs 20 (Android 13)
No Google -> Official Freestyle Libre App doesn't open
Juggluco Version: 8.0.11

j-kaltes, 2024-04-29:
Did you try hard enough to scan?

If you have root, you can better hide it for Juggluco and also hide Magisk.

Schabolon, 2024-04-29:
Thanks a lot for the fast reply.
I tried to scan at least 30 times over the course of 3 hours, and always get the same error message.
My phone is not rooted.

j-kaltes, 2024-04-29:
You can run a logging version from https://www.juggluco.nl/Juggluco/download.html#bugreport and send trace.log to my email address: jaapkorthalsaltes@gmail.com

Schabolon, 2024-04-29:
Thanks, I sent the log file.

j-kaltes, 2024-04-29:
With what app did you activate the sensor with that account ID?

Schabolon, 2024-04-29:
I followed the instructions from https://www.juggluco.nl/Juggluco/libre3/
The instructions said to choose a arbitrary account ID, so these are just numbers I entered manually.
I tried (after failing with my S10+) to use the sensor with my older phone and the Juggluco App, but the device is quite old and the app just crashed.

j-kaltes, 2024-04-29:
What phone was it? With what system ROM? Was it rooted? 
Did scanning succeed?

Schabolon, 2024-04-29:
The other phone is an Samsung Note 3 with LineageOS 18.1 (Android 11) with root.
Scanning didn't succeed, the phone kept vibriting for quite some time and nothing happened/the app crashed on different attempts.

j-kaltes, 2024-04-29:
Android 4.3 (Jelly Bean), upgradable to 4.4.2 (KitKat), upgradable to 5.0 (Lollipop) https://www.gsmarena.com/samsung_galaxy_note_3-5665.php
Maybe it has an old Linux kernel.

But you sensor seems to be activated.

Schabolon, 2024-04-29:
What exactly do you mean by `using a somewhat larger account ID will no longer by possible.`?

j-kaltes, 2024-04-29:
A sensor that is activated with a particular account ID can only be used with that particular account ID.

Schabolon, 2024-04-29:
Oh, I see.
So it could be that the "FreeStyle Libre 3, Scan error, Try again" message is due to the fact, that I use the wrong accound ID?

j-kaltes, 2024-04-29:
No. That is not what I meant. I want to say if the sensor is activated, you can't try another account ID.

Schabolon, 2024-04-29:
Should I try to get the official FreeStyle Libre 3 App to connect with the sensor, get the Account ID from there and than try to use Juggulco?
Or is that the wrong way?
Sorry, I am a quite confused as what to try next in order to connect with the sensor.

j-kaltes, 2024-04-29:
What I said also means that you can't use it with the official Freestyle Libre 3 app, because your account ID number doesn't correspond with a Libreview account.
You can try the patched one, because you can there enter the account ID manually. But why should that work when Juggluco doesn't?

j-kaltes, 2024-04-29:
No, enter in Juggluco a libreview account and use "Get Account ID" to get an account ID. This can also be used with the Abbott's  Libre 3 app from Google Play.

Schabolon, 2024-04-29:
When I try that, I get the following error:
```
ProviderInstaller.installIfNeeded:
dw
  at bf0.a(Unknown Source: 51)
  at tk.glucodata.Liebreview.libreconfig(Unknown Source: 52)
```

Schabolon, 2024-04-29:
Aside from the login (which is only relevant for the next sensor, as far as I understand) is there any way to "recover" a connection with my current sensor?

j-kaltes, 2024-04-29:
> I get `postgetauth: status=10 reason=invalidUserOrPassword` But I can login at https://www.freestylelibre.de/anmelden/login.html with these credentials.
Somehow you must have misspelled something.

Schabolon, 2024-04-29:
> > I get `postgetauth: status=10 reason=invalidUserOrPassword` But I can login at https://www.freestylelibre.de/anmelden/login.html with these credentials.
> 
> I am talking about Libreview.com not [www.freestylelibre.de](http://www.freestylelibre.de)

Oh, that explains it, I'll give it a try, thanks a lot.

j-kaltes, 2024-04-29:
> Aside from the login (which is only relevant for the next sensor, as far as I understand) is there any way to "recover" a connection with my current sensor?
Try the patched: https://www.juggluco.nl/Juggluco/libre3/patched.html

Schabolon, 2024-04-29:
When trying to scan the sensor with the patched libre3 app, I get the same error message.

j-kaltes, 2024-04-29:
The last option is to try it with another phone and when it works, send it via mirror to you current phone and thereafter remove the mirror connection and turn on use Bluetooth.

Schabolon, 2024-04-29:
Ok, I'll see if thats possible, I currently don't have a third phone at hand.
What would I be sending via mirror?

Venipa, 2024-04-29:
Ah I fixed it @j-kaltes 
I used a different android phone and transfered it by following the guide on the website, afterwards I was able to scan new sensors directly with the app

j-kaltes, 2024-04-29:
> Ok, I'll see if thats possible, I currently don't have a third phone at hand. What would I be sending via mirror?

If you want to keep the data on your current phone, you first mirror scans and stream to the other phone, then you reverse the direction of the mirror, sending Scans and Stream from the other phone to this phone. 
Hereafter you scan the sensor with that other phone. After receiving all data on your current phone. You turn off "Sensor via Bluetooth" on that phone and remove the mirror connection and turn on "Sensor via Bluetooth" on your current phone.
https://www.juggluco.nl/Juggluco/mirror/index.html

Schabolon, 2024-04-30:
Perfect, mirroring from another device finally worked :partying_face: 
Thank you very much for all the help!!