j-kaltes/Juggluco #153: Does Juggluco work on Android NFC tablet without a phone?

sshivaji, 2024-05-02:
I see many NFC Android tablets. Am curious if this in theory will work with juggluco and libre3 or would one still need an android phone?

j-kaltes, 2024-05-02:
One could do that.  After sending sensor information via a mirror to an Android tablet, you can also connect the sensor to a tablet without NFC, see https://www.juggluco.nl/Juggluco/mirror/index.html

sshivaji, 2024-05-02:
I saw that, it's a great feature! However, a mirror still needs a phone. Can people use the direct NFC functionality with an android tablet. For example. imagine someone has an iphone .. :)

j-kaltes, 2024-05-02:
Yes, as long as it is Android 8 or higher and has NFC, you can scan a Libre 3 sensor. I didn't try it , but it should be possible.

sshivaji, 2024-05-02:
Wow, great to note! This also can replace needing to buy a reader for those whose phones do not work with the libre 3 app for example.