j-kaltes/Juggluco #154: Scan error on FSL2-RU sensor change

AgushkaNightscout, 2024-05-03:
I have [same](https://github.com/j-kaltes/Juggluco/issues/41) issue with L2 RU version sensor. My dia-phone (Cubot Poket, Android 10) can't intercept already activated (by Realme 10) with Abbot's original app L2-RU sensor with NFC (error 18), although L3 could activate. But another phone (Realme 10, Android 13) can (no error)! I activate sensor with Realme and then transfer the sensor to Cubot using Juggluco's "Get Glucose values on Android device without NFC" greatest option. Why can this happen?

j-kaltes, 2024-05-03:
It is certainly not the same issue as you refer to. That issue was solved long ago with an update of Juggluco.

When you can't scan a sensor with a particular phone, there is something wrong with NFC of that phone.

AgushkaNightscout, 2024-05-03:
"I changed to a new libre2 sensor and it scans reliably in libre2 app, but juggluco always gets a scan error (18)." - I have exactly the same app behavior (version 8.0.3). Why does FS3 activate without problems but FS2RU does not activate (intercept) with error 18? Do they have such different NFC elements?

j-kaltes, 2024-05-03:
But you can scan the same Libre 2 sensor with Juggluco on another phone?

AgushkaNightscout, 2024-05-03:
Yeah, and it's totally confusing. Cubot can't even activate FS2RU with original Abbot's app. I'll try to take it apart and see what's wrong with the NFC module. But why FSL3 can be activated? I can't get my head around)

j-kaltes, 2024-05-03:
You can run a logging version from https://www.juggluco.nl/Juggluco/download.html#bugreport and e-mail trace.log to me. You can also send logcat.txt:
adb logcat -d >logcat.txt

If it is a US like Libre 2 sensor, the scanning is very complicated.

AgushkaNightscout, 2024-05-03:
Ok, thank you for advice! Sensor will expire in 5 days, then I will activate (intercept) a new one with the logging version and send you logcat.txt.

j-kaltes, 2024-05-03:
Why not just scan with you current sensor and send trace.log and logcat.txt?

AgushkaNightscout, 2024-05-03:
Oh, right! I forgot how cool your app is! When my kid gets home from kindergarten, I'll scan it.

gui-dos, 2024-05-03:
@j-kaltes:

> If it is a US like Libre 2 sensor, the scanning is very complicated.

Yep: according the attached screenshot that I received, it corresponds to the US 2 newer `2B` type: notice the `A` generation and the `08` regional code:

[image]

j-kaltes, 2024-05-03:
I looked at the logs. You scanned two times with two different errors. A scan consists of multiple NFC commands and some of them succeed and others failed. But NFC often fails momentarily because the sensor wasn't placed on the right spot of the phone or so, as was certainly the case with the first error. 
Because I have only one scan, I can't  judge whether scanning never comes past the point of the second scan error.

AgushkaNightscout, 2024-05-03:
I applied the phone a few times, all the time error 18, please take a look at logs on email.