j-kaltes/Juggluco #156: Differences between Widget-Arrow and Webserver-Service response?

nigzak, 2024-05-08:
I have build a simple app to read data from juggluco webserver API

I query against 
http://<IP>:17580/x/streams?token=<my_token>

example response:
'''
....
XX0F.......	13119	1715176336	2024-05-08-15:52:16	2	13118	157	-1	FALLING
XX0F.......	13120	1715176396	2024-05-08-15:53:16	2	13119	157	-0.97	STABLE
XX0F.......	13121	1715176455	2024-05-08-15:54:15	2	13120	156	-0.99	STABLE
XX0F.......	13122	1715176517	2024-05-08-15:55:17	2	13121	155	-0.99	STABLE
XX0F.......	13123	1715176576	2024-05-08-15:56:16	2	13122	154	-0.98	STABLE
XX0F.......	13124	1715176637	2024-05-08-15:57:17	2	13123	154	-0.94	STABLE
XX0F.......	13125	1715176695	2024-05-08-15:58:15	2	13124	153	-0.94	STABLE
XX0F.......	13126	1715176756	2024-05-08-15:59:16	2	13125	152	-0.86	STABLE
XX0F.......	13127	1715176815	2024-05-08-16:00:15	2	13126	151	-0.82	STABLE
XX0F.......	13128	1715176875	2024-05-08-16:01:15	2	13127	150	-0.82	STABLE
'''

I have observed that the widget shows of juggluco shows with this latest 150 value "slight down arrow"
Here in API response it shows STABLE
... should it not be similar to widget working (=FALLING)?

looking to documentation API response would be correct for "STABLE" (-1 ... 1 = STABLE)
https://www.juggluco.nl/Juggluco/webserver.html

but still there would be my question why is it different to widget? Has this another logic build in to calculate the arrows?
I would really like to have same arrow type as juggluco arrow if somehow possible ...

j-kaltes, 2024-05-08:
Juggluco doesn't use widgets to display glucose values. It displays the rate of change as as the the angle between the arrow and the x-as. It applies a threshold of 0.8 so that when there are only small changes, it displays an horizontal arrow. The direction of change labels in the last column are those of Abbott.

nigzak, 2024-05-08:
I was refering to this widget from juggluco which has a value and a arrow for trend

[image]

j-kaltes, 2024-05-08:
The glucose widget displays exactly the same arrow as in Juggluco itself.

nigzak, 2024-05-08:
I compared the API response (/x/stream) and the widget
they are showed in my case "STABLE in the API response" vs. "slight arrow down in the widget" ;)

I can also check if I find a similar situation again and provide screenshots, at the moment it is not the case and both are similar ("stable" and "arrow to right")

j-kaltes, 2024-05-08:
Look at the rate of change. TGhe rate of change -0.97 is displayed as a falling angle corresponding to -0.97, because of the threshold of 0.8. But an Abbott down arrow is only shown when the value is between -1 and -2.
Abbott has only a 5 arrows, Juggluco has all between stages as long as it is smaller than -0.8 or larger than 0.8

nigzak, 2024-05-08:
Hi,

that explains it, thanks.
=> CLOSED

closed 2024-05-08 by nigzak

j-kaltes, 2024-05-08:
In Juggluco 8.0.15, you can also specify a threshold of zero. That will make things clearer.
https://www.juggluco.nl/Juggluco/download.html

nigzak, 2024-05-09:
Hi again @j-kaltes 
thanks for the hint (I will not change it :) )

maybe I add some more explanation why I created this ticket in the beginning: I searched explicit the webpage and I did not find information about differences between arrow calculation of "orginal apps" and "juggluco" (as I wrote - maybe I did not find the correct subpage ...)

For me it is like I did never see other arrows than in the original app (straight, up, down, slight up, slight down) so I tought it would be "similar" (as you explained - it is not, juggluco has even "more arrows")
Mostly I have this arrows also in juggluco shown and if it is "more" (I see a huge down/up in the charts afterwards) I have a "sensor error" so I expect you have maybe other arrows which I don't see because my sensors saying then "no" :)

I also searched for the value you said here (0.8) in this ticket and also did not found any page for it.
At all I will for now stay on the text value which is responded via API in my application - still knowing that this is at least in my case currently not really good working - but I don't want to add a 3rd different arrow calculation in my own raspberry-pi app which confuses me maybe in months again because it is different (as example I know now -0.8 ... 0.8 - but also not the next gates you used without check the code or find the sub-webpage I did not find because I was too dumb :) ).

At all maybe it would be a good idea to add a general table (like the explanation of the API with "-1 .. 1 == STABLE") somewhere so it is totally clear that juggluco has another calculation in comparision to the original app (knowing that you don't have only ab*ott as source maybe not that easy ... don't know)

Don't see my ticket as "it is bad" - it was originally a question and maybe you could slightly improve the webpage adding this detail also there ... but I don't know, for me it is not a "must" because I know it now - but maybe for other ones (?)

Thanks again & have a nice day :)