j-kaltes/Juggluco #157: Problem Sibionics reading on app update

yiyoMX, 2024-05-09:
Hi, first of all thanks for your work on this app.
I've been using juggluco with Sibionics sensor and it was working great, until I updated the app. I was like 4 builds behind. I had a working sensor already when I updated, I had 115mg/dl but when I made the update it jumped to 1245 or something crazy like that, then it dropped to 250 and it settled somewhere around that number. I checked with a glucometer and I was 106.
What can I do? Sensor is like 20days old.

j-kaltes, 2024-05-09:
Juggluco test whether a value is valid: https://github.com/j-kaltes/Juggluco/blob/be880bee5c243b554aa14001ca09f9b5bc1a5d6f/Common/src/main/cpp/SensorGlucoseData.h#L115
Among which it checks that a value is below 502. Where did Juggluco display 1245? Have you found a place where this check is not made?
Such large values will be displayed as missing values.

yiyoMX, 2024-05-09:
On the juggluco app and it broadcasted it to aaps. I deleted it from aaps
for safety.
Also, I started another sensor because and it started displaying new values
since 1 min first was 45 then 180 and it jumped around for a little while,
now it seems stable, the problem is that this values where also
broadcasted, maybe it should have an option to disable broadcasts for x
time, while the new sensor warms up.

j-kaltes, 2024-05-09:
I should have corrected it strait away, but I thought it didn't matter because of the reason I just mentioned. A few values that are 1080 mg/dL too large is every well possible. But after 5 minutes the values are normal.

j-kaltes, 2024-05-09:
The bug should be fixed in 8.0.17: https://www.juggluco.nl/Juggluco/download.html
But for you it doesn't matter at all. The bug affects only the first four minutes with 8.0.16.

yiyoMX, 2024-05-09:
The thing is that it never went back to normal, it settled on 236 while I
was 106 with a glucometer, before the update it was showing 116. I waited
like 30 mins to see if it went back to normal but it didn't and decided to
change the sensor for a new one.
It's weird that it was that off with the update, while using the same
sensor.

j-kaltes, 2024-05-09:
That is something of the sensors. If you have a Chinese telephone number you can run the Chinese Sibionics app and receive all past values. https://cgm-prod-public.oss-cn-shenzhen.aliyuncs.com/app/update/android/%E7%A1%85%E5%9F%BA%E5%8A%A8%E6%84%9F.apk

When you press on Blood glucose device, followed by view data, you can view and save the individual values.

yiyoMX, 2024-05-10:
I dont have one.
So its better to update the app when the sensor time is over?

j-kaltes, 2024-05-10:
If you update from Juggluco before version 8.0.9, you can better update with a new sensor. From a later version is doesn't matter.

Geniy83, 2024-08-07:
Hello. Thanks for the great features of Juggluco.
I want to ask. Can Juggluco read European Sibionics sensors or only Chinese ones?
And is it possible to simultaneously use different libraries for different regions of the Sibionics sensor in Juggluco?

j-kaltes, 2024-08-07:
I only tested Juggluco with Chinese Sibionics sensors. I think that it will work exactly the same way with European Sibionics sensors.

Geniy83, 2024-08-07:
Russia began purchasing Sibionics from China. They are changing the packaging and I think they are using European sensors. In Russia they are called Hemotonix. Juggluco connected to this sensor but no data was sent. Data flows into the original application stably

Geniy83, 2024-08-07:
[image]

Geniy83, 2024-08-07:
And where can I find out the Status designation?

j-kaltes, 2024-08-07:
On your screenshot Juggluco is trying to connect to the sensor for only 2 minutes. You certainly have to wait longer. Status=19 means that the sensor hanged up, for example because it connects to another app.
Did the other app start again?

Geniy83, 2024-08-08:
I'll try to send you the logs today

cemcv25, 2024-08-14:
I have the same problem of connecting and failing to read Glucose data using the European Sibionics GS1 sensor. As suggested already, I will email you directly the trace.log

Geniy83, 2024-08-14:
Please send me this log file too. I myself did not use this sensor, I helped a person remotely, he was unable to retrieve the log file.
3477734n@gmail.com

j-kaltes, 2024-08-14:
> I have the same problem of connecting and failing to read Glucose data using the European Sibionics GS1 sensor. As suggested already, I will email you directly the trace.log

Geniy83  (https://github.com/j-kaltes/Juggluco/issues/157#issuecomment-2273841463) talked about sensors for the Hematonix app (https://play.google.com/store/apps/details?id=com.sisensing.rusibionics). He assumed it to be European Sibionics sensors. But why should that be so? You can use them for 16 days with Hematonix app, with the European app you can use sensors for only 14 days. Can you use European sensors for 16 days with the Hematonix app?
Now  cemcv25 (https://github.com/j-kaltes/Juggluco/issues/157#issuecomment-2288560118) talks about problems with European Sibionics GS1 sensors as if they are Hematonix sensors. With what app do these European Sibionics GS1 sensors work? With https://play.google.com/store/apps/details?id=com.sisensing.sijoy ?

cemcv25, 2024-08-14:
The official app I use is https://play.google.com/store/apps/details?id=com.sisensing.sijoy&hl=en_GB using the  Sibionics gs1 sensors sent to Europe from Hong Kong. Sorry for the confusion, but my question was based on  Geniy83 Initial post, where I understood there were two versions of the Sibionics gs1 sensor, Chinese and European. I was trying to understand why juggluco wasn’t connecting to my sensor.  I hope that clarifies my question and the reason I sent the trace.log file.

Geniy83, 2024-08-15:
They asked me for help connecting a Hematonix sensor to Juggluco. The person doesn't live nearby. Representatives sent the person 2 sensors, then called and told him not to pay attention to the box of the second sensor, that it was also Hematonix (photo below).
[image]
After looking at the photo, I concluded that Russia is simply changing the packaging of Sibionics EU.
When connecting to Juggluco, the Hematonix app was removed from the phone and the connection failed. The person couldn’t get the logs, there’s no experience, that’s why I didn’t email you anything.

Geniy83, 2024-08-15:
The Hematonix app was reading the second sensor (in the white box), but Juggluco was unable to connect. Because of this, I concluded that there are some changes in the library

j-kaltes, 2024-08-15:
Then it is the case that Juggluco only works with **Chinese** Sibionics sensors. I have modified the github description.

cemcv25, 2024-08-15:
Does the trace.log file that I sent to you confirm this ?

j-kaltes, 2024-08-15:
trace.log contains very little data. The five bytes the the sensor sends, fail on a validity check, which probably means that the sensor communicates differently. How I don't know.
Chinese Sibionics sensors also now and then did send data failing the validity check, but they only send other data which can be interpreted. Because so little data is available I don't know if that is also the case now.

Geniy83, 2024-08-15:
Is it possible to use the library in Juggluco for Sibionics EU?

j-kaltes, 2024-08-15:
The problem in trace.log has nothing to do with any library. You can use the libraries in the EU Sibionics app with Chinese Sibionics sensors.

Geniy83, 2024-08-17:
Hello. Let me send you a Hematonix sensor to test it?

woter86, 2024-08-27:
Hello! 
I collected some Bluetooth logs with this sensor. It seems you need four commands to run it and another prefix. You will need wireshark to view the log.
 [hematonix.log](https://github.com/user-attachments/files/16770261/hematonix.log)
[juggluco.log](https://github.com/user-attachments/files/16770262/juggluco.log)

Geniy83, 2024-10-20:
> I have the same problem of connecting and failing to read Glucose data using the European Sibionics GS1 sensor. As suggested already, I will email you directly the trace.log

Hello. If you currently have Sibionics EU installed, then have the Juggluco apk for this sensor.
[woter86](https://github.com/woter86) and I was able to connect the Hematonix and Sibionics EU sensor (for Europe) to the Juggluco. We can send the apk to your email.

j-kaltes, 2024-10-20:
If you modified Juggluco I am also interested in adding it to Juggluco. 
I haven't tried a EU Sibionic sensor myself because I expected it to be a lot of work, but I am still interested in adding modifications to make it work.

woter86, 2024-10-28:
> If you modified Juggluco I am also interested in adding it to Juggluco. 
> I haven't tried a EU Sibionic sensor myself because I expected it to be a lot of work and was busy with a Dexcom g7 sensor, but I am still interested in adding modifications to make it work.

Hi!
1. You can parse data from sensor with V120SpiltData:
  int[] arr_v1 = new int[2];
  byte[] arr_b2 = new byte[0x1C00];
  byte[] arr_b3 = new byte[2];
  boolean z = false;
  if (CGMDataHandle130.V120SpiltData(0, value, arr_v1, arr_b2, z, arr_b3, value.length) <= 0) {
    CGMDataHandle130.V120SpiltData(0, value, arr_v1, arr_b2, true, arr_b3, value.length);
  }
  
2. To launch sensor:
2а) Data for first command: 
private byte[] makeDescriptor(BluetoothGatt bluetoothGatt) {
    // Sibionics EU
        v120RegisterKey("56CE249349040C94F8B4B2375A8752D5CBE7A17814B502D9132489C0BFDFC99F0CAC670E8CBB085AF1C780B3D282E3");
        // Hematonix
    //v120RegisterKey("60B05FEB7C0A148DEED2B3375A8754D9D0E6A5751BCE02D9132489C0BFDFC99F0CAC670E8DA7115CEACF87B7DE8FD4612E1B7638C2");
        byte[] arr_b = new byte[0x400];
        String[] arr_s = bluetoothGatt.getDevice().getAddress().split(":");
        StringBuilder stringBuffer0 = new StringBuilder();
        int v2;
        for (v2 = arr_s.length - 1; v2 >= 0; --v2) {
            stringBuffer0.append(arr_s[v2]);
        }
        byte[] ad = convertDigitsToHex(stringBuffer0.toString());
        int v3 = CGMDataHandle130.V120ApplyAuthentication(1, true, 0, ad, arr_b, 0x400);
        byte[] data = new byte[v3];
        System.arraycopy(arr_b, 0, data, 0, v3);
        return data;
    }
2b) If reply to 2a: u16reply_ack_type == 0xC002
  byte[] arr_b3 = new byte[0x400];
  int v8 = CGMDataHandle130.V120Activation(0, true, new byte[2], ((long) (System.currentTimeMillis() / 1000L)), 1234, arr_b3, 0x400);
  data1 = new byte[v8];
  System.arraycopy(arr_b3, 0, data1, 0, v8);
  service2.setValue(data1);
  bluetoothGatt.writeCharacteristic(service2);
2c) If reply to 2b: u16reply_ack_type == 0xC008
  byte[] arr_b = new byte[0x400];
  int v1 = CGMDataHandle130.V120IsecUpdate(0, true, new byte[2], ((long) (System.currentTimeMillis() / 1000L)), arr_b, 0x400);
  data1 = new byte[v1];
  System.arraycopy(arr_b, 0, data1, 0, v1);
  service2.setValue(data1);
  bluetoothGatt.writeCharacteristic(service2);
2d) If reply to 2c: u16reply_ack_type == 0xC004:
  byte[] arr_b2 = new byte[0x400];
  currentIndex = Natives.getSIindex(dataptr);
  int v12 = CGMDataHandle130.V120RawData(0, true, new byte[2], currentIndex, 0, arr_b2, 30);
  data1 = new byte[v12];
  System.arraycopy(arr_b2, 0, data1, 0, v12);
  service2.setValue(data1);
  bluetoothGatt.writeCharacteristic(service2);
  
  
Answer to command:
String result = new String(arr_b2, StandardCharsets.UTF_8).trim();

{"u16reply_ack_type":49154,"u8reply_ack_resule":1,"u8error_code":0}

Answer with glucodata:
{"index":280,"temp":310,"current":84,"dump":2934,"reindex":0,"glouse":0,"trend":0,"gwarn":0,"twarn":0,"cwarn":0,"itime":1724775480}
temp and current must be pass to NativesAlgo (as Sibionics)

when can you do it?

j-kaltes, 2024-10-28:
Why don’t I receive the APK and why did I receive this information from you and not [Geniy83](https://github.com/Geniy83)?

I am now kind of forced to put it in Juggluco, but I still can’t do that correctly without more information.
    • To what characteristic is the output of makeDescriptor written? service1 or service2?
    • I need some logging of the values of the arguments before and after calling V120ApplyAuthentication, V120Activation, V120IsecUpdate, V120SpiltData and  V120RawData and their the return values.
      
For example CGMDataHandle130_V110SpiltData did nothing more then transforming the binary representation in a string of exactly the same information. For humans it becomes more readable, but for a computer it is more complicated and a waste of time.

Input: 
00 01 01 24 0B BA 00 33 00 2D 85 FF 00 B5 00 02 01 24 0B B9 00 33 00 32 85 FE 00 B5 00 03 01 25 0B B9 00 33 00 37 85 FD 00 B5 00 04 01 26 0B B9 00 32 00 3C 85 FC 00 B5 00 05 01 27 0B B7 00 32 00 41 85 FB 00 B5 00 06 01 27 0B B7 00 32 00 46 85 FA 00 B5 00 07 01 28 0B B7 00 32 00 4B 85 F9 00 B5 00 08 01 29 0B B6 00 31 00 50 85 F8 00 B5 00 09 01 28 0B B6 00 31 00 55 85 F7 00 B5 00 0A 01 28 0B B5 00 30 00 5A 85 F6 00 B5 EF 

Output: 
{"index":1,"temp":292,"current":51,"dump":3002,"status":45,"leftsend":34303,"nexttime":181},{"index":2,"temp":292,"current":51,"dump":3001,"status":50,"leftsend":34302,"nexttime":181},{"index":3,"temp":293,"current":51,"dump":3001,"status":55,"leftsend":34301,"nexttime":181},{"index":4,"temp":294,"current":50,"dump":3001,"status":60,"leftsend":34300,"nexttime":181},{"index":5,"temp":295,"current":50,"dump":2999,"status":65,"leftsend":34299,"nexttime":181},{"index":6,"temp":295,"current":50,"dump":2999,"status":70,"leftsend":34298,"nexttime":181},{"index":7,"temp":296,"current":50,"dump":2999,"status":75,"leftsend":34297,"nexttime":181},{"index":8,"temp":297,"current":49,"dump":2998,"status":80,"leftsend":34296,"nexttime":181},{"index":9,"temp":296,"current":49,"dump":2998,"status":85,"leftsend":34295,"nexttime":181},{"index":10,"temp":296,"current":48,"dump":2997,"status":90,"leftsend":34294,"nexttime":181}

The input is just an array of 16 bit big endian numbers with the same information. 
But this is maybe also changed.

Geniy83, 2024-10-28:
> Почему я не получаю APK и почему я получил эту информацию от вас, а не от [Geniy83](https://github.com/Geniy83) ?
> 
> Теперь я вынужден поместить его в Juggluco, но я все еще не могу сделать это правильно без дополнительной информации. • В какой характеристике записан вывод makeDescriptor? service1 или service2? • Мне нужно некоторое логирование значений аргументов до и после вызова V120ApplyAuthentication, V120Activation, V120IsecUpdate, V120SpiltData и V120RawData и их возвращаемых значений.
> 
> Например, CGMDataHandle130_V110SpiltData не сделал ничего, кроме преобразования двоичного представления в строку точно такой же информации. Для людей это становится более читаемым, но для компьютера это более сложно и пустая трата времени.
> 
> Ввод: 00 01 01 24 0B BA 00 33 00 2D 85 FF 00 B5 00 02 01 24 0B B9 00 33 00 32 85 FE 00 B5 00 03 01 25 0B B9 00 33 00 37 85 FD 00 B5 00 04 01 26 0B B9 00 32 00 3C 85 FC 00 B5 00 05 01 27 0B B7 00 32 00 41 85 FB 00 B5 00 06 01 27 0B B7 00 32 00 46 85 FA 00 B5 00 07 01 28 0B B7 00 32 00 4B 85 F9 00 B5 00 08 01 29 0B B6 00 31 00 50 85 F8 00 B5 00 09 01 28 0B B6 00 31 00 55 85 F7 00 B5 00 0A 01 28 0B B5 00 30 00 5A 85 F6 00 B5 EF
> 
> Вывод: {"index":1,"temp":292,"current":51,"dump":3002,"status":45,"leftsend":34303,"nexttime":181},{"index":2,"temp":292,"current":51,"dump":3001,"status":50,"leftsend":34302,"nexttime":181},{"index":3,"temp":293,"current":51,"dump":3001 ,"status":55,"leftsend":34301,"nexttime":181},{"index":4,"temp":294,"current":50,"dump":3001,"status":60,"leftsend":34300,"nexttime":181},{"index":5,"temp":295,"current":50,"dump":2999,"status":65,"leftsend":34299,"nexttime":181}, {"index":6,"temp":295,"current":50,"dump":2999,"status":70,"leftsend":34298,"nexttime":181},{"index":7,"temp":296,"current":50,"dump":2999,"status":75,"leftsend":34297,"nexttime":181},{"index":8,"temp":297,"current":49,"dump":2998 ,"status":80,"leftsend":34296,"nexttime":181},{"index":9,"temp":296,"current":49,"dump":2998,"status":85,"leftsend":34295,"nexttime":181},{"index":10,"temp":296,"current":48,"dump":2997,"status":90,"leftsend":34294,"nexttime":181}
> 
> Вход — это просто массив 16-битных big endian чисел с той же информацией. Но это может быть также изменено.

Now I’ll send you the APK by email. Woter86 contributed to this. He did more of it. He will send you additional information. We just took a lot of your code from native

woter86, 2024-10-28:
[call_log_d2.txt](https://github.com/user-attachments/files/17546383/call_log_d2.txt)

[image]

j-kaltes, 2024-10-28:
In the log file you send, I miss the content of the arguments after the function returns.
For example of V120SpiltData you log:
```log
[WARN] + Call: public static native int com.no.sisense.enanddecryption.CGMDataHandle130.V120SpiltData(int,byte[],int[],byte[],boolean,byte[],int)
[WARN]          - Args[0]: 0
[WARN]          - Args[1]: 60,-1,109,-112,8,59,-115,-34,-38,-74,-121,-115,4,-5,6,-116,-126,-124,-27,-36,41,-49,97,-74,126,10,-6,120
[WARN]          - Args[2]: 0,0
[WARN]          - Args[3]: 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0, etc
[WARN]          - Args[4]: true
[WARN]          - Args[5]: 0,0
[WARN]          - Args[6]: 28
[WARN]          - Return: 2
```
But the content of the arrays changes in the function. Before calling the function they are empty and afterwards some are filled.

I also need output of the Override functions in  SiGattCallback.  In Debug and Logging builds most are logged to /data/data/tk.glucodata/files/logs/trace.log. onCharacteristicChanged is logged in a native function that is removed, so it should be added, something like:
```java
byte[] value = bluetoothGattCharacteristic.getValue();
showbytes("onCharacteristicChanged "+bluetoothGattCharacteristic.getUuid().toString(),value);
```
You can then send trace.log. You get it out of the device with:
```bash
adb shell run-as tk.glucodata cat /data/data/tk.glucodata/files/logs/trace.log > trace.log
```

woter86, 2024-10-28:
> In the log file you send, I miss the content of the arguments after the function returns. For example of V120SpiltData you log:
> 
> ```
> [WARN] + Call: public static native int com.no.sisense.enanddecryption.CGMDataHandle130.V120SpiltData(int,byte[],int[],byte[],boolean,byte[],int)
> [WARN]          - Args[0]: 0
> [WARN]          - Args[1]: 60,-1,109,-112,8,59,-115,-34,-38,-74,-121,-115,4,-5,6,-116,-126,-124,-27,-36,41,-49,97,-74,126,10,-6,120
> [WARN]          - Args[2]: 0,0
> [WARN]          - Args[3]: 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0, etc
> [WARN]          - Args[4]: true
> [WARN]          - Args[5]: 0,0
> [WARN]          - Args[6]: 28
> [WARN]          - Return: 2
> ```
> 
 You can call this funtion to check result. I have no more logs.

woter86, 2024-10-28:
> ```java
> byte[] value = bluetoothGattCharacteristic.getValue();
> showbytes("onCharacteristicChanged "+bluetoothGattCharacteristic.getUuid().toString(),value);

This data you can find in wireshark log https://github.com/user-attachments/files/16770261/hematonix.log

j-kaltes, 2024-10-28:
Ok, how do I find the value in onCharacteristicChanged etc in the output of hematonix.log?

j-kaltes, 2024-10-28:
But, I can also find it in the input of V120SpiltData.

j-kaltes, 2024-10-28:
Is hematonix.log made at the same time as [call_log_d2.txt](https://github.com/user-attachments/files/17546383/call_log_d2.txt) so that I can search for the same data? For example for 3C FF 6D 90 08 3B 8D DE DA B6 87 8D 04 FB 06 8C 82 84 E5 DC 29 CF 61 B6 7E 0A FA 78 (the hex representation of 60,-1,109,-112,8,59,-115,-34,-38,-74,-121,-115,4,-5,6,-116,-126,-124,-27,-36,41,-49,97,-74,126,10,-6,120 that was the second argument to V120SpiltData that came from bluetoothGattCharacteristic.getValue() in onCharacteristicChanged)

woter86, 2024-10-29:
I have no more logs. My code in comment work and args is constant.

woter86, 2024-11-01:
Hi! Is there any forecast for the timing?

j-kaltes, 2024-11-01:
You can try a logging version https://drive.google.com/file/d/11gYlUFN_kZgUvIX8bfM60gO8T-AWYptQ/view which is absolutely untested.
It has application id: tk.glucodata.debug so that it doesn't damage you current Juggluco installation.
After trying it you can e-mail trace.log to me.
You can get it out of the device with:
```bash
adb shell run-as tk.glucodata.debug cat /data/data/tk.glucodata.debug/files/logs/trace.log > trace.log
```

Geniy83, 2024-11-02:
> You can try a logging version https://drive.google.com/file/d/11gYlUFN_kZgUvIX8bfM60gO8T-AWYptQ/view which is absolutely untested. It has application id: tk.glucodata.debug so that it doesn't damage you current Juggluco installation. After trying it you can e-mail trace.log to me. You can get it out of the device with:
> 
> ```shell
> adb shell run-as tk.glucodata.debug cat /data/data/tk.glucodata.debug/files/logs/trace.log > trace.log
> ```

Hello. Here is the log you asked for
[trace.log](https://github.com/user-attachments/files/17606180/trace.log)

j-kaltes, 2024-11-02:
Yes, it didn't work because it contained a little mistake. Is this one better:
https://drive.google.com/file/d/1Qm2vNauL3d-tXrt8CWZqjqpmrUQZYjjH/view?usp=drive_link

Geniy83, 2024-11-02:
When I connected the sensor, I quickly noticed that it connected and the application immediately crashed. Then I went into the application again and there was no connection there anymore. I'm sending you a new log
[trace.log](https://github.com/user-attachments/files/17606480/trace.log)

j-kaltes, 2024-11-02:
Thanks, If the following version crashes again, logcat is also useful:
https://drive.google.com/file/d/1DhEDoYz5bpUKfK6a8086GFlhh_Zyxj_X/view

Geniy83, 2024-11-02:
Loading of data packages from history failed
[Uploading trace.log…]()

Geniy83, 2024-11-02:
[trace.log](https://github.com/user-attachments/files/17606607/trace.log)

Geniy83, 2024-11-02:
I need to go away on business. I can help you in 2 hours

woter86, 2024-11-02:
Do you call methods for hematonix from native?

j-kaltes, 2024-11-02:
A new attempt:
https://drive.google.com/file/d/1VqMoDpr8d0Ysy26S3RMqgxwpOTNPjGnk/view

Geniy83, 2024-11-02:
not successful (
[trace.log](https://github.com/user-attachments/files/17607114/trace.log)
((

Geniy83, 2024-11-02:
Jaap, would you like https://github.com/woter86 to help you? He edited your code. You have a lot of native. Can you push your code for viewing?

j-kaltes, 2024-11-02:
New version:
https://drive.google.com/file/d/1B5DEyJEQsQWyziv-RY9OA1QNPosfuI_C/view?usp=drive_link

Someone said he was going to send me some EU Sibionics sensors, then I can test it myself.

Geniy83, 2024-11-02:
I offered to send you a Himatonix sensor, but you remained silent and did not answer.
I tested this version - it didn’t work (((
[trace.log](https://github.com/user-attachments/files/17607630/trace.log)

Geniy83, 2024-11-02:
[woter86](https://github.com/woter86) 
knows what commands to send to the sensor, there are 4 of them

j-kaltes, 2024-11-02:
Yes, I had already used a Dexcom g7 sensor and thought that I could better finish that before beginning with something new. But now I decided to do EU Sibionics sensors first. If I really receive that EU Sibionics sensor, the sensor from you is probably not needed anymore.

All four commands are present in the last log you send me:
siAuthBytes
siAsknewdata
getActivation
V120IsecUpdate

New version: https://drive.google.com/file/d/16A6KEWzdjJCORfxJUOxzBSodSvDiNLWU/view

Geniy83, 2024-11-02:
didn't go well (((
[trace.log](https://github.com/user-attachments/files/17607994/trace.log)

Geniy83, 2024-11-02:
[image]

[41 more replies]