j-kaltes/Juggluco #159: 8.0.19 and libre 3

onne111, 2024-05-11:
I updated  to 8.0.19 and my libre 3 who is on from 28.04 and officialy and expected date is 12.05  changed in  expected end date 4.05.2024 after  juggluco update to vers 8.0.19 ,no official date expire.
sensor still working,i put in paralel libre 2 ,started today,everything is ok,start date 11.05,expected and offic date 25.05.

I dont know what to think,libre 3 is extended or what?

j-kaltes, 2024-05-11:
A bug in 8.0.19 only. It should solved in 8.0.20. It will be available within 10 minutes at https://www.juggluco.nl/Juggluco/download.html

onne111, 2024-05-11:
updated to 8.0.20 and fixed.In vers 8.019   libre 2 is extended or just libre 2 plus?

j-kaltes, 2024-05-11:
Previously 14 days was hard coded. Now it uses the wear duration as received from the sensor for Libre 2 sensors. This will only make a difference for Libre 2 Plus.

closed 2024-05-11 by j-kaltes