j-kaltes/Juggluco #165: List window: Tapping on an item sometimes opens an edit window for the wrong item.

shaneharper123, 2024-06-01:
Tapping on an item in the List window will open an edit window for the wrong item when the first item displayed doesn't appear at the very top of the left column.

Below is an example of where the first item displayed doesn't appear at the very top of the left column.

[image]

I pressed the `<-` button multiple times until my first item was visible. (Pressing the `<- `one more time then results in the first item being displayed at the very top of the left column.)

Tapping on an item opens the edit window for the "following" item when the first item is displayed one position down from the top. (Tapping on the empty space at the top of the left column opens the edit window for the first item.)

The issue could possibly be resolved by ensuring that the first item displayed in the List window always appears at the very top of the left column.

j-kaltes, 2024-06-01:
I never experienced that the first item was missing. Do you have any idea what caused it?

shaneharper123, 2024-06-01:
It **always** happens when I go to the first "page" by pressing the `<-` button several times.

I am running Juggluco v8.0.22.

j-kaltes, 2024-06-01:
Yes, I can now reproduce it. I will look at it.

j-kaltes, 2024-06-01:
Does this solve the problem: https://drive.google.com/file/d/1cOIkUcwuhb6rEX_C7fw_-ZzOql7QPBse/view ?

shaneharper123, 2024-06-01:
> Does this solve the problem: https://drive.google.com/file/d/1cOIkUcwuhb6rEX_C7fw_-ZzOql7QPBse/view ?

Yes. Thank you.

shaneharper123, 2024-06-03:
Fixed in v8.0.23.

closed 2024-06-03 by shaneharper123