j-kaltes/Juggluco #166: Server Web internet please ?

spordiab, 2024-06-11:
Hello everyone,
I saw in the options that Juggluco has an integrated Nighscout web server.

At home locally, I can access the phone's web server from the computer and it works perfectly.
I can interrogate JSON to get the information: for example the IP address of the phone: http://192.168.50.217:17580/api/v1/entries/sgv.json

Now I wanted to know how to access the phone's web server from the internet, for example when the phone is not local but in another city?

http://xx:xx:xx:xx:17580/api/v1/entries/sgv.json

If anyone can help me that would be great.

THANKS

j-kaltes, 2024-06-11:
You can either forward a port to Juggluco or [Juggluco server](https://www.juggluco.nl/Juggluco/cmdline/index.html). Read the manual of your modem for how to do that.

spordiab, 2024-06-11:
> You can either forward a port to Juggluco or [Juggluco server](https://www.juggluco.nl/Juggluco/cmdline/index.html). Read the manual of your modem for how to do that.

I don't think we're talking about the same thing.

My child is at school and I want to connect to their phone from home to view: /api/v1/entries/sgv.json

On my children's phone, there is Juggluco which communicates with its Freestyle 2 sensor. In the Juggluco options there is web server activated.

But I don't know how to do it because it has to go through the internet and not with the local IP address.

THANKS

j-kaltes, 2024-06-11:
Most mobile data connections don’t allow port access. For that reason you need a go between. You can for example leave a Android phone or tablet at home and send data from the phone to this device, as described at: https://www.juggluco.nl/Juggluco/phones/index.html or you can run Juggluco server on a computer connected to the internet and create a mirror connection to Juggluco server. Both can also be used as a Nightscout server.

spordiab, 2024-06-12:
> Most mobile data connections don’t allow port access from the internet. For that reason you need a go between. You can for example leave a Android phone or tablet at home and send data from the phone to this device, as described at: https://www.juggluco.nl/Juggluco/phones/index.html or you can run Juggluco server on a computer connected to the internet and create a mirror connection to Juggluco server. Both can also be used as a Nightscout server.

Thank you for your answer;

it works well when I use an Android tablet as a gateway ^^.

Now I'm going to try to find a tutorial to make a web server directly with a raspberry pi 3B, it would be more discreet than a tablet.

If you have any other advice to give me, don't hesitate.

Thank you so much