j-kaltes/Juggluco #173: Notifications not working with Xiaomi Smart Band 8 Active

hierr, 2024-07-05:
Hi, first of all thanks for the work, I've just started using it since Libre 2 Plus was launched in my country.
Unfortunately it is not being able to redirect notification through the official Mi Fitness app to my Xiaomi Smart Band 8 Active. I've set up all permissions and notifications settings up, with no luck.

hierr, 2024-07-05:
On another note, it would be nice if to have (maybe on top of the most recent glucose reading) a metric like time in range and average glucose for the day, maybe not only for the day but even on a customizable time frame. Also, i felt the need of a way to lock the zoom so i don't accidentaly change my current view of X hours.

j-kaltes, 2024-07-05:
Maybe Gadgetbridge (see left menu->Watch->Help) works with _Xiaomi Smart Band 8 Active_: https://gadgetbridge.org/basics/topics/xiaomi-protobuf

In Juggluco left middle menu->Statistics gives time in range and average glucose for a customizable number of days.