j-kaltes/Juggluco #174: Build error in Android Studio Jellyfish | 2023.3.1 Patch 2

ArunM03, 2024-07-10:
I am facing build error while setting up this code in my Android studio. 
Its primarily related to CMake.

Android Studio Details : 
Android Studio Jellyfish | 2023.3.1 Patch 2

Execution failed for task ':Common:buildCMakeDebug[arm64-v8a]-10'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `C:\Users\ADMIN\AndroidStudioProjects\Juggluco\Common\.cxx\Debug\2a5r5v1s\arm64-v8a'
  [1/35] Building CXX object CMakeFiles/g.dir/net/sendcommands.cpp.o
  [2/35] Building CXX object CMakeFiles/g.dir/net/watchserver/sslserver.cpp.o
  [3/35] Building CXX object CMakeFiles/g.dir/share/logs.cpp.o

j-kaltes, 2024-07-10:
The log messages you show don't display an error.

ArunM03, 2024-07-10:
What do you mean , can't I build the project under windows ?

ArunM03, 2024-07-11:
I have compiled the project without any error.
I have installed the apk which is generated from that source code.

But when I try to scanning its giving this error. 

[image]

Please help me to resolve this

j-kaltes, 2024-07-11:
Again:  https://github.com/j-kaltes/Juggluco/blob/primary/README.md