j-kaltes/Juggluco #183: FR: have phone and watch automatically hand over connection to the sensor

StefanSarzio, 2024-08-01:
As long as phone and watch have an active IP connection to each other they should be able to determine which one connects to the sensor on their own, without human intervention. 

In general I want to phone to handle the connection to the sensor. 

But I leave my phone behind on several occasions. Right now for each of these occasions I have to switch the sensor connection twice. 

It would be a great improvement if upon losing the connection to the sensor the phone would check with the watch weather the watch can connect to the sensor. Then the mirroring should be reversed. 

The phone should still try to connect to the sensor and upon having a connection again tell the watch to stop connecting to the sensor and reverse the mirroring.

Puderzo, 2024-09-12:
If the original suggestion is not feasible, could it at least be possible to do the handover without having to have both devices present in the same network (Wifi/BT)? We have an LTE watch, and it maintains the IP connection just fine with the phone when taken far enough apart that the LTE kicks in, but unless they are in close proximity, the connection cannot be handed over (for example, when my son forgets his phone at home and leaves with just the watch, I can't hand the sensor over to the watch manually, but need to call him on the watch and have him activate the sensor from there (resulting in a "we both see the sensor" -situation when he gets home, and the phone sees the sensor again, as the mirror function has not been reversed)).

rbvitrb, 2024-10-01:
Hello, one question i can not get to work the direct sensor watch connection, once i establish it i get status error 133, can you please tell me what watch and what wear os and juggluco version are you using? Thank you!

StefanSarzio, 2024-10-01:
Pixel Watch 2, Wear OS 4 and Juggluco 8.0.23

rbvitrb, 2024-10-01:
And you only check the direct sensor watch connection button and works?

lev-n, 2026-10-14:
This comment proposes a concrete algorithm for the request in this issue, which asks for automatic handover but does not yet say how the devices would decide and switch. The feature would be opt-in: a new receiver mode that is **Off by default**, so nothing changes unless the user selects it. Nothing is implemented; I would write the code only if you agree with the approach.

Code references are to `primary` at 11.4.1 (`a983788a`); `cpp/` = `Common/src/main/cpp/`, `java/` = `Common/src/main/java/tk/glucodata/`.

## Summary

- **Modes:** Off (default, exactly today's behaviour) / Auto / Force phone / Force watch.
- **The phone is the arbiter:** every planned decision (including whether low glucose holds it), every tie-break, the low-battery plan. The watch decides only for itself: rule 5 (phone unreachable), its orphan scan, its claim at its own first reading, resume after a restart.
- **Ownership** is a persisted epoch (phone even, watch odd, so they never collide). The watch yields only to a phone decision; a higher epoch alone never makes a device stop.
- **A planned switch is release-before-acquire** with a sync barrier, acknowledgements and rollback, over a new Wear message path `/rswitch`.
- **Low glucose blocks planned switches:** while glucose is at or below the low-alarm level, or projected to reach it within 30 min, and the current receiver's connection is stable, no automatic planned handover starts, in either direction. Force retries included. With a watch receiver, the watch's own verdict counts too, and a lagging phone reading cannot clear the hold. Rules that act only when nobody is receiving are not blocked. The user's own manual switches are never blocked, but warn.
- **Existing mechanics are reused:** `Applic.setbluetooth`, the mirror row direction (flipped locally), `switchSync`, and `afterUnpair` with a result check. The automatic path never sends `/netinfo` ownership changes, `/bluetooth`, `sBlueWatch` or `/unpair`.
- **One pure C++ function decides:** `rs::decide(inputs, state, params) -> {actions, nextState}`, with one host test per row of a 197-row scenario table.
- **Scope:** the full design below, about 2,090 production LOC and 1,400 LOC of host tests (§10).

Why a new mechanism: the existing switch is fire-and-forget on the phone (`java/Applic.java:1089-1101`) and needs a sync under 180 s old (`cpp/watch/netinfo.cpp:984-987`), so it fails exactly when a device has been left behind.

## 1. Inputs

| Input | Source | Fallback / notes |
|---|---|---|
| Battery % | sticky `ACTION_BATTERY_CHANGED` (both) | Peer value older than 15 min = unknown: no new plan; an existing rule-6 plan is kept. |
| Charging | `EXTRA_PLUGGED` + dynamic `ACTION_POWER_(DIS)CONNECTED` (both) | 10 s debounce. A charging watch counts as off the wrist. |
| On wrist | `TYPE_LOW_LATENCY_OFFBODY_DETECT` (API 26), while `keeprunning` runs (watch) | OFF must hold 30 s, ON 60 s. Unknown → worn unless charging (`assumeWornWhenUnknown`). |
| Watch drain | least-squares slope over 120 min (span ≥ 20 min, drop ≥ 2 %) | EWMA, then 6 %/h. |
| Own-BLE reading | `SuperGattCallback.handleGlucoseResult`, current readings only | The **only** evidence that commits or claims ownership. |
| Missed readings (`missedPct15`) | owner: share of the expected own-BLE readings missed in the last 15 min (15 expected for 1-minute, 3 for 5-minute sensors), from the same hook, every sensor type. The window starts no earlier than the moment this device became owner; a window shorter than max(5 min, 2 × interval) counts as 0 % | Only the owner has it; the phone gets the watch's value in STATUS. |
| Mirrored reading | new value accepted in `processglucosevalue` | Counts for "is anybody receiving"; never commits (it may be an echo). |
| Proximity (rule 2 only) | `Node.isNearby()` or a ready BleMirror link | Unknown → not nearby. |
| Bluetooth, power saver | `ACTION_STATE_CHANGED`, `isPowerSaveMode()` (both) | Bluetooth off or watch power save → incapable. Phone power save → phone degraded. |
| Glucose, rate, low-alarm level | side-effect-free read of `lastValidStream()`; the device's own low-alarm level | No reading for 30 min → ELEVATED, and LOW_OR_LOW_SOON is false. |
| Sensors | active types, warm-up, end of life, NFC start / Libre 2 rescan; Libre 3 RSSI; Accu-Chek bond | – |

| Derived signal | Definition |
|---|---|
| Fresh | A reading within `freshSec` = 2 × interval + 60 s: 180 s (Libre, Sibionics, AiDEX X), 660 s (Dexcom, Air, Accu-Chek). Own and mirrored tracked separately. |
| Link UP / DOWN | UP if an `/rswitch` message arrived within 180 s. Otherwise, when a rule needs it, a **probe round**: 3 probes at 0/15/30 s, alternating carriers; all fail (≥ 45 s) → DOWN. Never shortened. Mirror traffic while DOWN = DATA_ONLY: no planned moves; mirrored readings still block unilateral rules. |
| Mirror in sync | A complete sync within 180 s. |
| Capable | Bluetooth on ∧ (watch) no power save ∧ battery above critical (watch 5 %, phone 3 %). |
| Watch eligible | Auto ∧ capable ∧ worn ∧ not charging ∧ ≥ 30 % (> 20 % if already owner; ≥ 25 % if the phone is degraded) ∧ time to 5 % unknown or ≥ 120 min (> 60 if owner) ∧ no backoff/penalty ∧ sensors allow it ∧ no warm-up, end of life or post-NFC hold. |
| Phone degraded | (≤ 15 % ∧ not charging) ∨ battery saver; clears at ≥ 25 % or charging, saver off. |
| Agreement | Ownership: same epoch and owner. Full: also same config hash, mode and sensor set. |
[... cut ...]
| 4 | Does a full Libre 3 pairing on one device invalidate the other's `kAuth`? Does Libre 2 enforce a monotonic unlock counter? How many simultaneous centrals or bonds do G7, Sibionics, Accu-Chek, Air and AiDEX X allow? Does a G7 re-bond show a pairing dialog? | – |
| 5 | Auto keeps `keeprunning` running on the watch (off-body events reach only foreground apps). Acceptable for battery? | yes |
| 6 | Experience with `TYPE_LOW_LATENCY_OFFBODY_DETECT` and `Node.isNearby()` on Galaxy and Pixel watches? | – |

nleva, 2026-10-14:
Hi, @j-kaltes 
I really appreciate your review and suggestions for this topic.
I tried to cover the algorithm of automatic switch and found an alternative approach - notification with a switch button in case of missed readings on a mirror.

P.S. I use OnePlus Watch 3

lev-n, 2026-10-14:
Hi, @j-kaltes
I'd really appreciate your review and suggestions on this.

While working through the edge cases, I also found a much simpler alternative: when the mirror misses readings, show a notification with a **"Switch receiver"** button. The user moves the sensor connection with one tap, and the app never decides on its own. It would reuse the existing manual switch and avoid most of the risks of the automatic version (both devices connected at once, devices disagreeing about who is the receiver, reconnect cost). However it doesn't cover some edge cases e.g. switch device at night in case of missing readings.

P.S. I use OnePlus Watch 3

j-kaltes, 2026-10-15:
I don't read this because it is too long, but I am not going to add automatic switching of connections.
It is not needed and it will certainly give problems:

- How can they communicate this change when they are out of reach?
- They will constantly switch with a simple connection error.

It is just a very stupid idea.

lev-n, 2026-10-15:
Thanks for the clear answer. I understand you won't add this, so I won't open a PR.
 Short answers to your two points, for anyone who finds this issue later:
- **Out of reach**: then nothing is switched. A planned switch only starts while both devices can talk to each other, and every step is acknowledged; if a step times out, the old receiver takes the sensor back. A device connects to the sensor on its own only when it has had no glucose at all, neither from the sensor nor through the mirror, for a set delay of a few minutes. That is what you would otherwise do by hand. When the devices are in contact again, the phone decides which one keeps the sensor.
- **Constant switching**: a single connection error doesn't trigger anything. A switch needs a condition that lasts, such as readings missed over 15 minutes, battery level or charging. After a switch the sensor stays put for a minimum time (an hour for Libre 2), there is a backoff after a failed switch, and a growing penalty if the watch keeps losing data. Low glucose, or glucose expected to go low within 30 minutes, blocks automatic switches while the current connection is stable.

 It is off by default, and with it off Juggluco behaves exactly as it does today.
 I'll keep it in my own build. Thanks for Juggluco.

j-kaltes, 2026-10-16:
When it takes to long before automatically making a connection with the sensor, you can better do it by hand at the moment you know you forgot your phone.
It has also the problem that when the connection between watch and phone doesn't work, so the watch doesn't receive glucose value from the phone, it will start trying to connect directly with the sensor so interfering with the connection of the phone with the sensor. 
I really see only problems with this automatic switching, and I see no reason why you shouldn't do it by hand.

closed 2026-10-16 by j-kaltes

lev-n, 2026-10-16:
Understood, thanks for explaining. For anyone who finds this later, these are the two cases where switching by hand doesn't help me:

- At night, when the watch runs flat while I sleep, nobody is awake to switch, while the phone next to the bed could have taken over.
- I only notice that the phone or the watch is missing after the closed loop has already stopped for lack of glucose.

I'll keep it in my own build - successfully tested today a few scenarios.

j-kaltes, 2026-10-16:
Trying to switch when both are within reach of the sensor, seems to me an even worse idea.
You create connection problems by letting multiple devices connect to the same sensor, you do not solve it that way.

StefanSarzio, 2026-10-16:
My two cents: I wanted to have this functionality because I often leave my phone behind.

Now with the latest technology, it is no longer a battery problem leaving the sensor connected to the watch all the time, so my use case has been solved.

lev-n, 2026-10-16:
Good to hear your case is solved, @StefanSarzio.

@j-kaltes is right: if a message is lost or the link drops during a handover, both devices can end up trying to connect to the sensor for a few minutes. The design makes that rare, not impossible. In my test on a phone and a watch today it did not happen.

The algorithm as an interactive simulation (the same code as in my build): https://sensor-switch-simulator.web.app/#/simulator

[image]