j-kaltes/Juggluco #185: Can't get 1-minute data from the application

KytGluc, 2024-08-06:
I have successfully activated the Libre3 sensor with juggluco, get glucose values in the app every minute. But when I try to export the data in tsv format, I get an empty file. what could be the problem?

j-kaltes, 2024-08-06:
Is the file completely empty or does it contain the header (_"Sensorid	nr	UnixTime	YYYY-mm-dd-HH:MM:SS	TZ	Min	mmol/L	Rate	ChangeLabel"_)?
Did you scroll the screen so that the last received glucose value was present on the screen? Did you specify a large  enough number of days and press _Stream_?

What happens when you turn on left menu->Settings->Web server and go to http://127.0.0.1:17580/x/stream?header&days=1 in chrome on the same phone as Juggluco is running?

KytGluc, 2024-08-06:
Yes, the file is completely empty. 
All glucose values i can see on the screen and graph, and i can export data from the xDrip, but only 5-min.
I can't access to http://127.0.0.1:17580/x/stream?header&days=1 in chrome...

j-kaltes, 2024-08-06:
If you can't access the webserver, you probably didn't  turn it on. Did you set [x] Active? What about the other commands from https://www.juggluco.nl/Juggluco/webserver.html?urlstart=http://127.0.0.1:17580   ?
What happens when you export other data with left middle menu->Export. For example _History_ or after adding an amount  to the  app, _Amounts_?
When you pressed Stream, did you come  into a Storage Access Framework dialog in which you pressed  _save_? After returning to Juggluco, was "_Saved_" written in the Export dialog below "_Scans  History  Stream_"?

KytGluc, 2024-08-07:
1. If you can't access the webserver, you probably didn't turn it on. Did you set [x] Active?   - Yes, I set [x] Active
2. What about the other commands from https://www.juggluco.nl/Juggluco/webserver.html?urlstart=http://127.0.0.1:17580 ?   - I can't get access to other commands.
3. What happens when you export other data with left middle menu->Export. For example History or after adding an amount to the app, Amounts?   - All files are empty.
4. When you pressed Stream, did you come into a Storage Access Framework dialog in which you pressed save? After returning to Juggluco, was "Saved" written in the Export dialog below "Scans History Stream"?    - Yes, I come into a Storage Access Framework dialog, but after returning I see "cancelled"

j-kaltes, 2024-08-07:
Where do you see _cancelled_? At the place I am talking about _cancelled_ is never written.
 [image]

In the picture at the place of 'Not saved"

KytGluc, 2024-08-07:
[image]

j-kaltes, 2024-08-07:
_Cancelled_ can only come from pressing left menu->Sibionics with the Export dialog open followed by the back button.

KytGluc, 2024-08-07:
I don't use Sibionics at all...

KytGluc, 2024-08-07:
Can I put another version of the app and connect an already activated sensor (6 days out of 14) to it? Or could this cause the sensor to fail?
Can I connect the sensor to the Juggluco app on another phone?

j-kaltes, 2024-08-07:
You don't need to use Sibionics sensors to press on Left menu->Sibionics.
What do you mean, "the sensor to fail"? Where you where talking about had nothing to do with failings sensors as far as I can see.

KytGluc, 2024-08-07:
I mean that I don't press Left menu->Sibionics at all.

The question is could i connect my activated Libre3 to another phone with another version of Juggluco?

j-kaltes, 2024-08-07:
That is a totally different issue. The answer is that you can as long as Juggluco uses the same Libreview account id number.

KytGluc, 2024-08-07:
I’ve updated version of Juggluco (8.0.4 ->8.1.5) and all is ok