j-kaltes/Juggluco #188: Improvements for using multiple sensors at once

sshivaji, 2024-08-10:
I suspect it might be common to want to use more than one sensor. Juggluco is so good that anyone using the stock libreview app clearly does not care about their health.

Anyway, I want to report a few issues when using 2 sensors, 2 different people.

1. On my Samsung watch, I sometimes get my blood sugar, sometimes my partners. It would be nice to have a tagging feature where I can tag my sensor and that of my partner. Failing that, can we lock only one sensor reading to the watch?
2. The same tag can help differentiate the 2 sensors when looking over it in the graph.
3. The Statistics are now incorrect. I would like stats for myself and for my partner separately, including average glucose/A1c etc. I am not sure what stats are being shown now.

Other than this, I think juggluco is amazing even if you have multiple sensors!

j-kaltes, 2024-08-11:
You shouldn’t use one exemplar of Juggluco with the sensors of multiple people. Instead run a second exemplar of Juggluco with a different application ID. The application ID should be the same on the phone and the watch. See: https://www.juggluco.nl/Juggluco/dub/index.html
You can use floating glucose to view them both at the same time.
(It is also possible to install multiple exemplars of the same app in other ways, for example with multiple Android users. I don’t know if that is possible on under WearOS).

sshivaji, 2024-08-11:
Wow, I will try that! Thanks.

sshivaji, 2024-08-12:
I still think it is useful to have 2 sensors of different people on one page, especially if it is a diabetic and a non-diabetic to be able to compare values, especially if the diet is similar. It was quite revealing with Juggluco even if it was not meant for that purpose. In fact, it is quite good even now, except the stats being incorrect. However, we can calculate the stats manually using the downloaded data.