j-kaltes/Juggluco #191: Support for FL3 Plus?

blaqone, 2024-08-17:
Here FL3 Plus is available. Is there anything known if the sensor works with juggluco?

https://www.freestylelibre.de/produkte/freestyle-libre-3-plus-sensor.html

j-kaltes, 2024-08-17:
Nothing is known.

gui-dos, 2024-08-17:
The recent update of the the Libre 3 app for the Italian market to the version number 3.6.0 (i.e. https://www.apkmirror.com/apk/abbott-diabetes-care-inc/freestyle-libre-3-it/freestyle-libre-3-it-3-6-0-release/freestyle-libre-3-it-3-6-0-android-apk-download/) may be related since the packaging is trilingual.

uschindler, 2024-08-20:
The only difference is that it works 15 days instead of 14 days. Some German Insurances already order them, mine (TKK) is still listing only Libre 3.

I think once somebody has the hardware available, we can test it with the app. Hopefully they only changed a few things, but in worst case it uses different encryption keys. BUT: There was no update of Libre 3 app until now, so it could be that the sensor protocol is unmodified except end date.

gui-dos, 2024-08-20:
@uschindler:

Notice that the last week there was indeed a version bump to 3.6 for the Italian Libre 3 app both for iOS and Android, i.e. https://www.apkmirror.com/apk/abbott-diabetes-care-inc/ (the packaging is trilingual DE/FR/IT, in fact).

blaqone, 2024-08-21:
I ordered one yesterday and will report.

uschindler, 2024-08-21:
Should there anything change for Libre3 without plus, because diff of new version has some analysis of wear duration, so I expect it to have a slightly different guess on end date when I install it now.

The Libre3 I had always survived a few minutes longer than it told?

uschindler, 2024-08-21:
Ok after scanning classical Libre3 it calculated same end time. So works fine. Let's see what reporter says.

gui-dos, 2024-08-21:
@uschindler:

Yep, if you stop the "Trident" app some minutes before the 14-day formal expiration you can notice that the current Libre 3 keeps on advertising for another whole day, then its status switches from `05` (expired) to `06` (terminated) and the sensor shuts down by itself.

Otherwise, as soon as you run "Trident" on the 15th day, you can sniff that the app is sending a specific "shutdown" control command and the sensor is not detectable via BLE anymore.

For the Libre 2 I noticed that the custom command `1A 1D` also disable the sensor Bluetooth but it can be reactivated countless time with the successive NFC subcommand `A1 1E` ;-)

blaqone, 2024-08-27:
With recent update FL3+ works perfectly fine.