j-kaltes/Juggluco #193: Uploading to Nightscout 15.0.2 not work

Stan-Di, 2024-08-23:
While testing uploading from Juggluco, I found out that uploading works fine in Nightscout 14.2.6, but does not work in Nightscout 15.0.2 - error java.io.FileNotFound ***/api/v1/entries

How this can be fixed?

Stan-Di, 2024-08-23:
I tested "test v3" in Juggluco and it also working in Nightscout 15.0.2! - https://termux-ssh.99ns.ru/

Stan-Di, 2024-09-13:
I spent a lot of time on tests, but I never got a result. But I was confused that one of my users has everything working.

I continued to search for the problem today. At first I tried disabling https, but that didn't help either. I tried changing all sorts of passwords and addresses on my server - it didn't help either. I tried to change the telecom operators - and it didn't help.

But now I tried to enter the Nightscout address by ip (setting up Nginx of course) and everything worked!

I repeat once again: at the same time, xDrip and AndroidAPS work by domain name!

I do not know how to fix it and do not understand the error java.io.FileNotFoundException but I think there is some problem in the code or in the library... Why is domain name access not working?

If you need additional debug logs, I am ready to provide

Stan-Di, 2024-09-13:
[image]

Screenshot

j-kaltes, 2024-09-13:
The uploader to Nightscout is developed for 15.0.2. I can produce your error message by misspelling the api-secret.

Stan-Di, 2024-09-13:
I did not change the api_secret when I made access by IP, not by site name. I think the problem is in DNS, because I gave the API secret and site name to another person and everything worked! I can also give you the address for testing. But unfortunately it does not work for me and another person from another country. If you tell me how to make logs, I will provide them to you

j-kaltes, 2024-09-13:
I fix only BUGS in Juggluco.

Stan-Di, 2024-09-14:
Then please tell me what libraries are used for the connection? Could they be the problem? Do they require Google Services?

I now tried to establish a connection again from a device on Android 8.1 and there is no connection even via IP and local network. Of course, I copy the address and api_secret from Juggluco and access this data via Chrome. I had to install xDrip and then everything worked. But I would like to exclude xDrip

I'm trying v1 and v3 again now.

[image]
[image]

pepkolajk, 2024-11-15:
doesnt work on my side either, tested url and api secret on postman just to be sure.

output on uploader: status `nothing `with string "Tried nothing"

it only works if i press test v3 which of course its gonna throw an error because I am using a V1

j-kaltes, 2024-11-16:
You probably didn't check "_Active_":
[image]
V1 does work. You can also upload with V3 when you use V1 clients, you only shouldn't switch between uploading with V1 and V3 without removing the old data.
[image]

pepkolajk, 2024-11-16:
Just created an new instance, fresh, with fresh database. Tested on web and postman all working fine.

Try to upload on juggluco, nothing.

[image]

Have tried everything/scenario.

j-kaltes, 2024-11-16:
Strange!
Did you ever switch away from Juggluco or press "_Send now_"?
Did you force stop and restart Juggluco?

pepkolajk, 2024-11-16:
- Yes
- nothing
- Yes, every scenario

j-kaltes, 2024-11-16:
When there is no new data in Juggluco, it keeps saying "Tried nothing". Is there new data in Juggluco? Did you press "Resend data"? What when you set "Send amounts" and enter some numbers in Juggluco?

Stan-Di, 2024-11-16:
> Just created an new instance, fresh, with fresh database. Tested on web and postman all working fine.
> 
> Try to upload on juggluco, nothing.
> 

So far I have no problems with Juggluco and the watch on Android 8 and it communicates with my Nightscout.

What I did:
1) I turned off the enhanced power saving mode
2) I disabled Google services kill mode

I think that the download from Juggluco uses libraries from Google and there may be some problems. Although they didn’t appear when xDrip and AndroidAPS were unloaded...

PS One of my users encountered problems uploading to Nightscout when he went on vacation to another region :)

jylilov, 2025-08-01:
I was also getting an `error java.io.FileNotFound ***/api/v1/entries`.

After debugging, I found out that this error appears when Nightscout API returns a non-200 HTTP status code. 

In my case, I was getting a 413 HTTP response code from Nightscout. I have Nginx behind Nightscout, which by default allows request body sizes only below 1 MB. When I just set `client_max_body_size 100M;` in the Nginx config, the issue was resolved. Maybe it can be helpful for someone.

But at the Juggluco side, it would be nice to have at least a bad response status from the Nightscout API in the log instead of a FileNotFound exception.

jrdeutsch, 2025-09-10:
Hi @Stan-Di - I'm experiencing the exact same problem, but I'm clearly not as technical as you, so I don't know how to fix my issue!  

I'm getting the dreaded "java.io.FileNotFoundException: https://myURL/api/v1/entries" error in Juggluco, and just a few days ago, things used to be fine. The issue seemed to coincide when I moved Juggluco to a brand new phone, but that may just be a coincidence.  

My Nightscout version is 15.0.2 and is hosted on Google Cloud.  I unfortunately don't know what Nginx is or how I would use it with Google Cloud, or even if it's necessary. 

Any advice? This issue between Juggluco and Nightscout is creating all kinds of logistical issues for me!

Thx.
Jon