j-kaltes/Juggluco #203: Libre 3 not showing values

imbahzor, 2024-09-09:
Hello,

I have a Libre 3 bought in norway, i am of course lucky enough to also have a samsung that doesnt connect to the device with the libre3 app, so i scanned it with juggluco and the sensor did start the 60 minute startup timer, but after it is showing no values.

i get this error: Not Ready Status 3. and then randomly it will start the 60 minute timer after a day or two again.

j-kaltes, 2024-09-09:
Do you mean,  "_X not ready. state=3_"?
That message comes from the sensor. I never received it. The sensor is probably broken and should be returned to Abbott. Can't you somehow test the sensor with Abbott's Libre 3 app to see what it says?

imbahzor, 2024-09-09:
> Do you mean, "_X not ready. state=3_"? The state information comes from the sensor. I never received it. The sensor is probably broken and should be returned to Abbott. Can't you somehow test the sensor with Abbott's Libre 3 app to see what it says?

Yes, tested with Libre 3 app on another phone and just get told to check the sensor so probably a broken sensor.

gui-dos, 2024-09-09:
They are not obfuscating the sensor states definitions ;-)

```
package com.adc.trident.app.frameworks.mobileservices.libre3;

/* loaded from: classes.dex */
public class MSLibre3PatchStates {
    public static final int PATCH_STATE_ERROR = 7;
    public static final int PATCH_STATE_ERROR_TERMINATED = 8;
    public static final int PATCH_STATE_EXPIRED = 5;
    public static final int PATCH_STATE_INSERTION_DETECTION = 2;
    public static final int PATCH_STATE_INSERTION_FAILED = 3;
    public static final int PATCH_STATE_MANUFACTURING = 0;
    public static final int PATCH_STATE_PAIRED = 4;
    public static final int PATCH_STATE_STORAGE = 1;
    public static final int PATCH_STATE_TERMINATED_NORMAL = 6;
}
```

j-kaltes, 2024-09-09:
> They are not obfuscating the sensor states definitions ;-)
>
> ```
> package com.adc.trident.app.frameworks.mobileservices.libre3;
>
> /* loaded from: classes.dex */
> public class MSLibre3PatchStates { [...]

Yes, it could very well be the same state as the MSLibre3PatchStates members refer to: With state==4 it is also working properly.