j-kaltes/Juggluco #204: Support for Stelo by Dexcom

kanbock, 2024-09-18:
Is there a plan to add support for this new no-prescription-required [Stelo by Dexcom](https://www.stelo.com/)?

It also seems like Stelo app can export data to Health Connect - is there a way to feed this into Juggluco?

I've used Juggluco with FS3 and absolutely love it over the standard app. Due to Stelo being the first to be sold OTC, it is a good alternative to more expensive options like Libre FS3 and others.

georgios1963, 2024-12-08:
It would be really great if someone could be working on this. If you need any resources let me know

chemputer, 2025-01-29:
I just ordered a couple Stelo biosensors, I'm not sure the update time, but some initial research shows that if you use Dexcom Clarity, it'll sync the data, and can be exported as a CSV (likely via an activity call if necessary), which Juggluco could then read to import new values. Alternatively, and perhaps more sanely, the [Dexcom API](https://developer.dexcom.com/support/faq) is an option, getting data through the API might be the best option if the first option doesn't work, or to perhaps in addition to it. It seems fairly simple. It requires OAUTH 2.0, I am fairly confident I can implement all that behind the scenes, but admittedly GUI stuff is not my forte (not that the app is exactly the best UI/UX but yeah) so I may need some help with that part. Unfortunately, to quote the webpage:

> The Dexcom Web API is a retrospective API with a three-hour delay. For us to consider your request for access to production data, we require that developers first create an app and test the app with sandbox data. If you have not yet created and tested your app, please do so before applying for an upgrade.

Great, three-hour delay. Still useful, to some extent, but not great.

They should be here in ~3-5 days, but if anyone would like to confirm using the Dexcom API that the data from Stelo (I honestly don't see why it couldn't) can be retrieved via the API, that'd be useful, though you'd need to register as a developer, then create an app, apply for limited access, then run the API command to auth and then grab the data. So it may be a bit to ask. 

One thing that may be an issue is Dexcom's process for going from Limited Access (less than 20 users) to Full Access, and the requirements involved, see [here](https://developer.dexcom.com/docs/dexcom/scopes-access/). It's a bit extra.

On the plus side, it would not just work with Stelo, it'd work with all Dexcom devices, so there's that. 

So, two options, potentially: 

- Brute force, but seemingly immediate: Use Android activity/intent to export CSV, then load that into Juggluco, check for new entries. (I have no clue how to implement that on iOS)
- Use Dexcom API, more work, may not be immediate, says up to 3 hour delay, though. 

I'm admittedly unsure if the API gets to the final app whether it should be under my developer account, or if someone more senior on the project should register, use their oauth credentials, and get the app approved for commercial use. My gut says the latter.