j-kaltes/Juggluco #205: Status bar overlaps settings window and makes units unclickable

rinukkusu, 2024-09-20:
I'm just installing the Juggluco App on my Xiaomi Poco and the unit selection renders under the status bar. Even with the "notch" turned off, where apps should render below the status bar and not "edge to edge", it still renders under the status bar.

[image]

Any ideas on how to get me set up? Can I maybe edit the settings manually somewhere in a file or database?

Thanks and BR

j-kaltes, 2024-09-20:
You should be able to scroll the screen down, so that the unit is below the status bar:
[image]
Scroll->
[image]

Do you use Juggluco 8.1.7?

rinukkusu, 2024-09-20:
Hi, this was the default when I installed the app afaik, literally just set it up. I was not able to pull it down from there, but after a few restarts and forth and backs, the app suddenly rendered below the status bar. So maybe this was just a MIUI quirk. No idea.

Thank you for looking into it, though!

closed 2024-09-20 by rinukkusu