j-kaltes/Juggluco #206: Can not establish direct sensor watch connection

rbvitrb, 2024-10-01:
Hello, new diabetic type 1 since July here.
I am trying to set the direct sensor watch connection following the instructions in the Juggluco webpage but I can not establish connection. I have my phone watch juggluco working, showing value on watch but once I check the direct sensor watch connection option y always get Bluetooth status error 133 on watch, that apparently means (as I read in juggluco help) that hasn't found any device to connect.
I have tried putting off bt on my phone, switching off phone, forgetting sensor on phone, switching off and on bt on watch and nothing works. 
Info: 
Smartwatch: Xiaomi smartwatch 2
Wear os v: 3.5
Juggluco watch v: 8.1.7
Juggluco phone v: 8.1.7
 Would appreciate some tips on the matter thank you!

j-kaltes, 2024-10-01:
There are some other actions you didn’t mention:
    • Reboot the watch;
    • Turn on Turbo;
    • See if you have given Juggluco on the watch all the needed permissions.

Can you show all the information under left menu→Sensor on the watch in a few screenshots? Maybe I see something remarkable.

rbvitrb, 2024-10-01:
It worked mate! I first switched off bt of phone and after 5 min I switched on phone, and it started working! 
Maybe I was to impatient for not waiting long enough or just the switch off phone. 
I even send back to the phone the data now.

Thank you mate!!! Have a nice day :)

closed 2024-10-02 by rbvitrb