j-kaltes/Juggluco #207: Sensor initialized via Juggluco

rbvitrb, 2024-10-04:
Hello, I have initialized my new libre 2 plus via Juggluco and when trying to scan the sensor with Libre view app i can only see the immediate value, it doesn't get the BT connection and therefore, I can only see the value when I scan NFC. 
With Juggluco i stablish connection and i see every min GC changing but with the previous sensor I could have it in both juggluco and libre view app.
I do receive data in libreview account for my doctor because I register it in juggluco app but I would like to have Libre view app running.
[image]
[image]
As you see in the image, it doesn't appear the real time value of the sensor.
I believe is due to the fact that i initialized the sensor with Juggluco instead of Libre view app. 
Has anyone any clue of how to get the BT connection back with Libre view app? Even though I started it with juggluco

Thank you

j-kaltes, 2024-10-04:
It is because you started the sensor with Juggluco. Librelink only uses Bluetooth when it starts the sensor itself to prevent conflicts with Freestyle Reader.
Anyway you shouldn't connect two apps at the same time with the sensor, because glucose alarms become less reliable: https://www.juggluco.nl/Juggluco/index.html#multiple

rbvitrb, 2024-10-04:
Okay I see, so for the next sensor I should always initialize with Libre view, thank you

j-kaltes, 2024-10-04:
You should always start a sensor with Juggluco, because otherwise Librelink will also try to connect with the sensor via Bluetooth and possibly disrupt the connection.