j-kaltes/Juggluco #208: Notification not working

Richspiral, 2024-10-08:
[image]

Since the recent update the notification is not working properly. As you can see in the screenshot the font is in the background and not properly visible.

This did happen with the previous version but I could tap on it, go thru to the app and it would adjust to be visible. That is no longer possible. 

The only way to get it to work now is to go into my notification settings and turn the notification on and off again.

For info my phone switches to dark mode in the evening and light mode during the day and it is when these modes are switched that the notification becomes much less visible.

Thank you.

j-kaltes, 2024-10-08:
Yes, when you switch from dark mode to light mode, the notification is momentarily invisible. The background changed color, but the notification is not rewritten. But when a new glucose value arrives it becomes visible again.

Richspiral, 2024-10-08:
Thanks for your prompt response. 🙏

The notification is not becoming visible again. It's nearly 2pm here in the UK and day mode has been active for many hours with my numbers rising. 

It's still invisible.

Even with the old version i had to click thru to the app to get visibility. That method is no longer working.

Richspiral, 2024-10-08:
Yep that works if I force stop the app and start it again, as does the method I outlined in my original post of turning on and off the notification.

j-kaltes, 2024-10-10:
Now I also see it. It will be fixed in the next version.

Richspiral, 2024-10-11:
Thanks for your efforts and work as always 👍