j-kaltes/Juggluco #210: Libre 3.Scan Error (Pixel 9, Android 14)

jmandel, 2024-10-18:
I just tried connecting with a new Libre 3 sensor on my new phone.

### Juggluco details 
Version Code: 480
Version Name: 8.2.2-phone-li3-si
arm64-v8a
Build time: 09:49:24 08-Oct-2024

### Phone details 
Model: Pixel 9 Pro XL
Android version: 14
Build number: AD1A.240905.004

I get the following error; when I try again there is no change. I have also force stopped the app, cleared app data and cache, and reinstalled with no change in behavior. 

[image]

I'm not sure how to generate better debugging information, but happy to take any steps that would be helpful.

j-kaltes, 2024-10-18:
Are you certain it is the phone and not the sensor? What is the expiration date of the sensor?
Running a logging version of Juggluco and e-mailing trace.log to jaapkorthalsaltes@gmail.com, could be of help as described here:
https://www.juggluco.nl/Juggluco/download.html#bugreport

jmandel, 2024-10-19:
I tried to confirm that the sensor was OK by installing the Abbott Libre 3 app; that app did successfully NFC recognized the sensor and started its measurement period.

When I paused the Abbott Libre 3 app and re-ran Juggluco, then Juggluco NFC recognized the sensor and seems to be working (still counting down the 60min waiting period).

So... it's working now but I failed to generate good error logs for you :/

j-kaltes, 2024-10-19:
So it isn't a new sensor type that doesn't work with Juggluco. But it is still possible that it is a new type that can't be activated by Juggluco. How often did you try to scan before starting this issue?

jmandel, 2024-10-19:
I tried ~10 times over ~5 hours to start from Juggluco before (successfully) starting from the Abbott app.

jonjonr6, 2025-04-08:
I have exactly this issue using a new Pixel 9 Pro (not XL) running Android 15.

Libre 3 starts and scans and operates perfectly from my Pixel 5 without any need of the Abbott App.

After 20 or so attempts to start a new sensor with the Pixel 9 Pro, I gave up and started it on the first NFC scan using the Abbott App. It subsequently worked OK with Juggluco over Bluetooth but multiple atempts to NFC scan resulted in the error as shown above. I tried repeatedly and eventually got a correct NFC scan from Juggluco "Sensor ready for use, add sensor end date to calendar app". It seems to take 10 or 20 attempts to get this to work. However, Juggluco can scan and read my Novopen Echo+ with NFC without any difficulty and the phone doesn't have problems with NFC scans using other apps.

I also find that the Bluetooth connection seems much less reliable with the Pixel 9 Pro compared with my Pixel 5.

Anyway, I need Juggluco more than I need a new phone so I'm going to send the Pixel 9 Pro back for a refund! I might try again in a few months if anybody reports any subsequent improvement. I'm also happy to provide any further info if needed.

jrdeutsch, 2025-04-08:
FYI - I have a Pixel 9 Pro running Android 15.  I have also had problems
with initial scanning, but after 20+ attempts, it did end up working.  And
I've experienced no issues with ongoing connectivity with Juggluco on my
P9P.

frmaas, 2025-04-20:
I experienced the same scan error today with a FSL3 sensor on a Pixel 8 with Android 15. I started the sensor using the FSL3 app, force closed the app, and juggluco would say for many tries that there was a scan error. Trying again often, resulted in success now :)