j-kaltes/Juggluco #213: Watch to sensor connection

greenie95125, 2024-10-25:
I'm pulling my hair out here. For months, I've been able to have my Galaxy Watch 6 Classic (and Watch 4 before that), connect directly with my Libre 3 sensors (with floating glucose), until last week. If I didn't want my levels sent to my provider via libre view, I wouldn't even have mirrored my levels to my phone. 

Now, no matter what I try, I cannot get my watch to connect reliably. I have no problem getting my phone (Pixel 7 Pro) to connect and send my values to the watch, but as soon as I try to go the other direction, I get "no connection" on the watch and of course the "no values from mirror" on the phone. Today when messing with it, it worked for about 15 minutes, and I thought I solved it, but it just reverted back to the errors above. 

On the watch menu item, I have clicked "default", init watch app and of course the direct watch connection box when I can. I've tried deleting the connections in the mirror menu on both the watch and phone, and started from scratch, nothing seems to work. I thought that it may have been the Android 15 update, but realized that this problem started before the update. I read on here that someone was having issues with watch faces other than Juggluco's, so I tried that to no avail. 

After using Juggluco for a year or so, I thought I finally understood how to set it up,  but I guess not. I can however, get it working on the phone with ease. Does anyone have any suggestions, or point me to a step by step guide for setting it up on the watch? I'm using 8.1.0 on the watch and phone.  Thank you for your help.

j-kaltes, 2024-10-25:
If you do receive glucose values on the watch when the phone is connected with the sensor, everything you talk about will not the be cause: the mirror connection and the Android version of the phone will not be the problem. Pressing default or "init watch app" will only cause more problems because all data is send again to the watch which takes time.
To find out what is wrong, you have to look at left menu->Sensor on the watch. Can you display it's content in a few screenshots?

Also the connection between watch and sensor is better when they are on the same arm.

greenie95125, 2024-10-25:
Thanks for the reply. Since last night I've had the sensor connected to the phone. Here is what the sensor screen looks like about 1/2 hr after switching it back.  I appreciate the help.
[image]
[image]
[image]

-Mike

j-kaltes, 2024-10-25:
Have you tried, when the sensor is connected with the phone, resend data in mirror? Or uninstalling the watch version, removing the mirror connection on  the phone and installing Juggluco on the watch again. And presing init watch app?

You can also run a logging version of Juggluco and send me trace.log,see https://www.juggluco.nl/Juggluco/download.html#bugreport

greenie95125, 2024-10-25:
I tried to resend data, and when I reversed the connection it seemed to work. I went out of range of my phone for a few min, and the level stayed on my watch. When I returned my phone said no glucose form mirror then shortly showed a value. Until abouy 10-15 min then back to "no value since:) on the watch and the mirror error on the phone. That has happened in a few of my previous attempts. This weekend, I will remove it from my watch and start fresh. The frustrating part is that it worked great for months. I even waited until I tried it with a new sensor before I reached out to you. Thank you again.

[1e21b293-6e82-4f4a-a24a-aa95bb67674e]

greenie95125, 2024-10-26:
I've reinstalled Jug on the watch with the same result. It's been over 30 minutes and while the spinner on the sensor menu shows the correct CGM, the rest of the entries remain blank, like it hasn't even tried to connect. On the phone, the direct d\sensor-watch box is checked. I can still (with ease) get the phone to connect and send the values to the watch. I'm not sure what else I can try. The watch shows sensor id: No Connection.

[6bc9ebcb-fd3c-425d-8b5b-e9a8b6231fdd]

greenie95125, 2024-10-27:
Well, today I uninstalled on both devices, then reinstalled (phone first). Same result, but then I read in one of you tutorials that other apps my interfere with the connection, and realized that the Libreview app was still installed, so I uninstalled it. Then I checked the direct sensor to watch button, and it looks like it's working again. Could that do it? It didn't affect the connection to the phone at all. 
Anyway, thanks for the help. and thank you for all the work you do to help fellow diabetics manage their disease! 

[3eadc118-6fb8-4194-934b-6461aa9dbeed]

greenie95125, 2024-10-27:
Here is the sequence of what happened I think. About 3 weeks ago, I started a sensor that scanned fine, but never sent any values. At first, I thought it was Juggluco, and started troubleshooting there. Obviously the watch wasn't getting any values, but when I tired to reverse the direction to the phone being connected to the sensor, that didn't work either. That was when I reinstalled the Libre app to see if that app could get a value, which also didn't work. At that point I decided it must be a bum sensor and started another. It also scanned fine, and immediately started sending values to my phone since that was how I had it set up at that point. Now out of sight out of mind is the Libre app which is still running in the background, and my saga of connecting my watch to the sensor began. 

I spent two weeks trying to sort this out when it was actually so simple. Thank you again. Also, is there anywhere I can make a donation to help with your development? I would love to contribute.

[ff2cabf1-b020-465d-b704-751b88b5e2e9]