j-kaltes/Juggluco #214: JUGGLUCO w/FSL3 errors messages

durgok, 2024-10-28:
For the last several months, quite often as soon as I activate the FSL3, it will work for few hours but then immediately go into sensor error and never revive. After some point, it goes to sensor stopped working with state = 7. This is for my 10 yr old daughter. I even tired updating to latest version for the last sensor which is when it moved from sensor error to sensor stopped working. Am very desperate cos my daughter is looping with AndroidAPS and very essential we have readings come and the sensor availability is very scarce these days. Please help me asap and ask me any questions that will help triage the issue. Her phone is Samsung galaxy S23fe. 

[image]
[image]

j-kaltes, 2024-10-28:
These are both problems with the sensor. A sensor error can be temporary, but when it takes to long you should get a replacement from Abbott. The state=7 message is send by the sensor to Juggluco and probably means that the sensor doesn't work anymore.
If Abbott's Libre 3 app is available in your country, you can get an error message in that app by  scanning the sensor with that app (after force stopping Juggluco.)

durgok, 2024-10-28:
Oh we can be very sure its a bad sensor causing this and not setting or set that does this? I want to eliminate any errors that I may be doing although I have been using it over 2 years now and started encountering these issues may be only in the last 6 months

j-kaltes, 2024-10-28:
Sensor errors and that state=7 have to do with the sensor, not with the reception of information from the sensor. It is the same for all apps and under Libre 2 both by scanning and Bluetooth.
But I consider it possible that how the sensor is place in the body can influence the reading. Maybe at some places it doesn’t make good enough contact with blood.

durgok, 2024-10-28:
Am trying to establish the pattern to see what is the problem as when I follow the same steps some goes just fine while others have the issue. Btwn I just tried to take the sensor over to Libre 3 by force stopping Juggluco. However Libre 3 app is not scanning it again, I mean when I hover over it, no vibration and nothing happens. Any thoughts about it?

kenkit, 2025-04-12:
just received that error, I think it happens when the readings go low for a long time. The sensor things it's faulty and just shuts down.
Not sure if recovery is possible.

kenkit, 2025-04-12:
Tried scanning with official app but it doesn't even scan, but errrors

kenkit, 2025-04-12:
a

durgok, 2025-04-12:
Mine sometimes do it right at the start of sensor too. At some point, it looked to me as if it’s a bluetooth issue. So I stopped having an overlap btwn two sensors. I terminate the existing one, make sure nothing shows up on the Bluetooth devices of the phone and then start a new one, all of these directly on Juggluco app without using the Libre 3 official app. Yes I wont have readings for an hour until new sensor warms up but incidentally the issue stopped(knock the wood) and I would rather wait an hour over losing my mind with this error on a presoaked sensor!

missmah, 2025-06-20:
I've just had this happen to the third sensor in a row! The latest one was only in for a few hours before this happened! It does seem to always be preceded by severe hypoglycemia..