j-kaltes/Juggluco #215: Menu display error

SimonXuku, 2024-10-29:
I use my Chinese smart watch DW99（android 8.1 arm64）,after lastest Juggluco installed,the menu display error.
menu is all black 
[image]

Juggluco 7.6.5 is ok.

[image]

j-kaltes, 2024-10-29:
Why is the expected end time of the sensor 01-02-2024? I also don't see the times at the top of the screen, the glucose and time lines and the low glucose line. Are they also missing? In the top screenshot the language is Chinese, in the bottom English, does the language setting matter?

j-kaltes, 2024-10-29:
Does this version solve the problem:
https://drive.google.com/open?id=17-UoEhn4Ktn9_2dlVHMMF5JOFXaXgwfs

SimonXuku, 2024-10-30:
It works fine now. thanks.
[image]