j-kaltes/Juggluco #217: Battery Drain

greenie95125, 2024-11-01:
After solving my sensor to watch connection issue (thank you), what I'm noticing now is a very fast battery drain. As you know, I completely removed and reinstalled the Jug app on my watch, so is there something I may have missed in the default settings when I reinstalled it? To be clear, I did not reset my watch, and all I did was uninstall and reinstall Juggluco. 

I used to get a full day with Jug running and only the BT radio on. Now I'm lucky if I get 12 hours. It's a problem because I sleep with my watch so, 6 or 7 hours is already used up by the time I wake up. Bedtime mode is on during sleep as well. I've even turned of heart rate monitoring, but that hasn't helped either. Thank you if you have any suggestions.

-Mike

j-kaltes, 2024-11-02:
When I use Samsung Galaxy Watch 4 only for looking at my glucose value the 100% battery is estimated to take more than 16 hours to get empty. The battery of Watch 6 classic is 425 mAh instead of 361 mAh, so it will take nearly (425/361*16=18.8) 19 hours before it is totally empty.
The energy consumed by receiving these glucose values from the sensor is constant and not very much. What cost energy is what you do with the watch: lighting up the screen and running other programs. When I record an activity the battery is empty in somewhat more than 3 hours. Also apps running in the background you not even use can consume battery power. You can best uninstall everything you don’t need.
https://xdaforums.com/t/galaxy-watch-6-7-adb-debloat-connecting.4613093/
[image]
Now the expected battery life is only 12 hour and 45 minutes. Maybe it is because yesterday the watch automatically recorded an activity.
The connection with the sensor is very good.

greenie95125, 2024-11-02:
Thank you for the kind & detailed response, I appreciate it. 

-Mike

https://xdaforums.com/t/galaxy-watch-6-7-adb-debloat-connecting.4613093/ [https://xdaforums.com/t/galaxy-watch-6-7-adb-debloat-connecting.4613093/]
battery.png (view on web) [https://github.com/user-attachments/assets/694f3ca7-0c7e-444a-a4e4-5ade8c6933a8]
[050d6989-eb80-4410-bac3-f6cadfd3a518]

j-kaltes, 2024-11-05:
I removed the Android checkbox, so your suggestions will be of little use. This means that it is always turned off. Letting Android make the connection would only substantially improve battery life when the watch often loses the connection with the sensor and the time to get connected could become worse.

greenie95125, 2024-11-05:
I'm not sure what to say. It has substantially improved my battery
performance. It's been 14 hours since it was taken off the charger, and I'm
at 73% now with 17 hours left. That's over  a 2x improvement over Juggluco
making the connection. I would ask you to consider leaving that as an
option. Yesterday I  changed the direction so the phone would send the
values because I didn't want to have to charge mid-day. I did not and have
not noticed any excessive sensor disconnects with either option.

greenie95125, 2024-11-05:
Not to be a pest, but it's been 18 hours since a full charge and I still have 62% (14 hours) left. With Juggluco managing the connection, I'd be charging my watch by now. The difference is actually astounding. Tomorrow, I will uncheck Android and post the results.

[4a1b60ca-e901-4bc9-b434-36ebbe226296]

greenie95125, 2024-11-07:
OK, this will be my last entry for this issue. Today, ith Juggluco managing the connection, at 18 hours I'm at 32% with 4 hours left, that's a remarkable difference compared to 62% with 14 hours left after 18 hours with Android managing the connection. At 20 hours (now) I just got the low battery warning at 15%. so I would have to charge before I leave work. 

So, I strongly suggest that you leave the Android option available and let the users decide. 

Finally, for these tests, all radios are off other than bluetooth. No wifi, location, or NFC. Thank you for your time and effort with this app, Jaap. It's a game changer for me and makes managing my disease MUCH easier!  -Mike

greenie95125, 2024-11-09:
Ok, last one I hope. Can you please tell me which in which version release you initially removed the "Android" check box? I didn't see it in your changelog. Thank you.

greenie95125, 2024-11-12:
?