j-kaltes/Juggluco #227: Juggluco and Health Connect - Sending Glucose Values Stopped and Resumed after Sensor Disconnected (On Different Days)

UberQueiroz, 2024-11-20:
Hello to all!

On november 17th my Libre 2 sensor got disconnected and after some reboots of the cellphone (Galaxy s23+ with Android 14) it got reconnected. But glucose values stopped being sent to Samsung Health through Health Connect. 

Health Connect was registering logs from Juggluco normaly, but apparently there was no glucose info on those logs and so nothing new appered on Samsung Health.

Until today when again the sensor got disconnected from Juggluco and a new reboot was required. After that new glucose values started to appear on Samsung Health - but all history from that interregnum wasn't imported.

Anyone knows why that happened?

I attached some screenshot to show what I described.

[image]
[image]
[image]
[image]

j-kaltes, 2024-11-21:
As I understand, it Juggluco sends properly to Health Connect, but values didn't end up in Samsung Health. So this is not a Juggluco issue, but something having to do with Samsung Health not receiving glucose values from Health Connect.

j-kaltes, 2024-11-21:
Samsung health is always behind the values in Health Connect, that has nothing to do with disconnecting.

mpirio, 2025-05-29:
Hi,

I'm experiencing the same issue. In my case, blood glucose values have never appeared in Samsung Health.

Juggluco is correctly connected to Health Connect — I can see the values it sends (see attached screenshot). However, absolutely nothing shows up in Samsung Health, even though the connection to Health Connect is active and I've granted all necessary permissions to Juggluco.

Any idea what might be going wrong?

Thanks in advance!

[image]
[image]
[image]

j-kaltes, 2025-05-30:
You need to give Samsung health allowance to access Health Connect  in "My Page"->Settings.
But it is not very useful. If an other app wants to use glucose values of Juggluco, it can better make use of the [glucose broadcast](https://www.juggluco.nl/Juggluco/glucosebroadcast.html) or [web server](https://www.juggluco.nl/Juggluco/webserver.html) in Juggluco.

j-kaltes, 2025-05-30:
The data format is determined by Health Connect. Health connect can interpret the data and send it further. It doesn't change from value to value. Samsung health receives these values, but it is late. Google Fit receives all values much earlier.
That is not an issue of data format.
It is possible that every minute is too much. I could send the history data to Health Connect instead of the stream values.
But heart rate is every second.