j-kaltes/Juggluco #230: Unable to find the libre 3 error code in juggluco

damgot, 2024-11-30:
This isn't really a problem, but rather a feature request (unless it already exists, in which case I haven't found it):

This is already the 2nd time my free sensor 3 has stopped prematurely (after 3-4 days instead of 14). It's obviously a problem with the sensor and not the Juggluco application. In the “Sensor” menu I can see that the “connection” keeps failing. The “error code” I have is “Status=133”. But I have no idea what this error code means. What's more, it doesn't correspond to the error code I get in events with the abbott application. 

Would it be possible to have the meaning of the Status numbers (not necessarily in the application directly but in a help menu or in an external documentation)?
Would it be possible to have the “Abbott” error codes directly in the application? Because every time I have to send my sensor back, I have to reinstall the “Libre 3” application from Abbott to be able to read the error code and enter it in the replacement form.

Thank you very much

j-kaltes, 2024-11-30:
A status=133 error message is mentioned in the help of left menu->Sensor, https://www.juggluco.nl/Jugglucohelp/sensorhelp.html
It happens very often and is normally not a reason to return a sensor. It doesn't correspond to an error code in Abbott's app, it is just not having a connection with the sensor.
To come to know what Abbott’s says, you have to try out Abbott’s Libre app and see yourself. You can pushing your findings on this github if you want.

damgot, 2024-11-30:
Hello.
Thank you for your reply. 
Unfortunately, my sensor has been in error for 15 hours now. Although I tried to restart my bluetooth connection several times, or even restart my phone completely. I also looked in the “Libre 3” application and I have a list of events bearing the number 57. And no blood glucose data.
So I put in a new sensor.
Surprisingly, however, when I scanned my new sensor (which was indeed a brand new one I'd just put in), the start-up time was 11 minutes in Juggluco and not 60 as usual... The 11 minutes have just elapsed (but not yet the 60 minutes) and I have a successful connection, a successful set-up but a failure on the glucose values (which I suppose is normal, given that the start-up time is normally 60 minutes).

j-kaltes, 2024-11-30:
If you had updated Juggluco to the latest version, it would have shown 60 minutes instead of 12.

damgot, 2024-11-30:
Oh yes, sorry. I updated on Monday to 9.0.0. I've just seen that we're already at 9.0.2. I'll update juggluco right away.
Thank you :)