j-kaltes/Juggluco #234: Total lost connection

Abdulvus, 2024-12-04:
Everything was working calmly when suddenly the sensor lost the connection and did not connect anymore. I tried to do a lot of things, even reinstalled the application to a new version, but it refuses to connect so far, although the usual scanning via NFC works.

n-dering, 2024-12-15:
i have the same problem, i've got notificaton from freestyle libre 2, but as soon as I scan with juggluco, I lose connection and I have to scan manually until I change sensor

j-kaltes, 2024-12-15:
This seems to be totally different issues. It is always the case that after you scan a Libre 2 sensor with Juggluco, Librelink misses the authentication information to receive glucose values from the sensor. But in this case Juggluco receives glucose values via Bluetooth from the sensor.  This is not a problem for people who want to use Juggluco.

The Abdulvus seemed to have a totally different problem. He didn't receive glucose values via Bluetooth in Juggluco. There are the following possible causes:
-  Librelink started itself and tries to connect with the sensor giving the wrong authentication information. Juggluco on another device or Freestyle reader can cause the same problem;
- Bluetooth on the device doesn't work anymore. This can sometimes be solved to turning Bluetooth off and on or rebooting the phone;
- There is sometime between the sensor and the phone or watch, for example your body.
See: https://www.juggluco.nl/Jugglucohelp/sensorhelp.html