j-kaltes/Juggluco #236: Will there be ever a Dexcom Clarity integration for Dexcom sensors?

twinko, 2024-12-10:
Hi @j-kaltes,

is it possible to integrate a Dexcom Clarity upload like for the libre to librelink? If yes it would be awesome if you could put this feature on your roadmap :)

Regards

itrn57, 2024-12-26:
I second this request. Jugcgluco is superior to the Dexcom app but doctors want to have access to BG data with Clarity. So it's imperative to integrate with Clarity.

MightTower, 2025-01-09:
Would also love that.

lostboy86, 2025-05-22:
Love the libreview function, hope the same awesome work @j-kaltes  do for the clarity option, expecially now after juggluco have full support of g7.

Good work!

lostboy86, 2025-05-22:
> For what do you need Clarity? Dexcom G7/ONE+ have backfill, so you can
> just turn of “*Use Bluetooth*” in Juggluco and start the official Dexcom
> app and let it run for 20 minutes to receive all data of the past 24 hours.
> Thereafter, you force stop Dexcom and turn on “Use Bluetooth”.

Official app are not supported on many phones, Oppo flagship, Motorola
flagship, phone update android 15,etc.

Diakem are based on 2 years old version of original app.

Most endos required data from Clarity.

These Is because i Hope for a Clarity integrations on juggluco.

User must remember every 8 hours or 24 hours tò stop jugglcuo, open diakem
(if also work at Launch of new g7 15days), backfill, stop diakem, reopen
juggluco.

Is the same problem of Libreview, many endos required official upload data

j-kaltes, 2025-05-22:
Yes, but submitting to Clarity isn't that easy either.

itrn57, 2025-05-22:
This is a brillinat idea to use Dexcom G7/ONE+ have backfill. 
Thank you !

j-kaltes, 2025-05-29:
As an alternative to using Dexcom Clarity, the webserver in Juggluco 9.7.0 can generate a report you can give your healthcare provider. See: https://www.juggluco.nl/Juggluco/webserver.html
When you have switched on left menu->Settings->Exchange data-> [Web server](https://www.juggluco.nl/Jugglucohelp/Nightscouthelp.html), you can also view the url in your browser via left middle menu->Statistics->Save.

j-kaltes, 2025-05-29:
What do you mean with "_accept only official data_"? You had an appointment with your endocrinologist and he refuses to listen or look at the data presented in any other way than Libreview or Clarity?
I just said to me endocrinologist that I didn't use Libreview and he looked in my app at left middle menu->Statistics to see my  percentage of time in range and percentage of time in hypoglycemia.
Don't your health ensurance pay your sensors when you don't use Clariry or Libreview?
I heard that in Poland they had to scan 8 times a day and they controlled that via Libreview. A lot of users asked me to created fake scans for that reason.

lostboy86, 2025-05-29:
Yes, my endo , but Is the same on many parts of Italy, not accept not
official datas.

If you not send the data trough official protocol, the prescription Is not
renew.

Same for show data on nightscout, tidepool, etc.

j-kaltes, 2025-05-29:
Can your endocrinologist and diabetic nurse constantly follow what you are doing via Libreview or Clarity? And when you don't agree with that they let you die?

lostboy86, 2025-05-29:
No, every 3 months retriever the sensors, and for this, if he don't read
all 90 days data on libreview or clarity, alert you if also some days are
missing data, and if you continue, you don't take the sensors

This Is official rules policy of dexcom or abbot, there are many doctors
outside Italy can admit users without this stric policy, but also in other
country, endos are the same of Italy, only official solution for datas.

For example doesn't care if you use juggluco or diakem, but datas must be
on libreview or clarity, for delivery the sensors.

I suspect this force for share datas users to Abbott and dexcom

j-kaltes, 2025-05-29:
I think they consider you immature.

lostboy86, 2025-05-29:
All Italy endo work like this.

Also, search only in many forums, endo wants data also with glooko, another
official solutions, and there are many people disperate in other country,
asking plugin from Nightscout to glooko, same libreview or Clarity.

Is not and endo decision, but a dexcom / Abbott contact : i sell the
sensors to tour doctors/ country/ medical hospital, if you pay these and
send me data of the users

j-kaltes, 2025-06-01:
Juggluco server 9.7.0-2 can save data in Libreview format. It also includes data from Dexcom and Sibionics sensors:
https://www.juggluco.nl/Juggluco/webserver.html#libreview
So if your doctor is really so stupid to accept Glooko’s but not Juggluco AGP, you can use it.

lostboy86, 2025-06-01:
He don't accept data in any format or other app, only in official system,
with diab id trough libreview or clarity.

vebo59, 2025-11-26:
Is it possible to send data from Dexcom G7 to Glooko like it is possible for Libre? That would be great,  specially when using  Omnipod 5. When a smartphone isn’t compatible with the official app from Dexcom, using  Juggluco would resolve the problem. Then endo can check the data in Glooko.

vebo59, 2025-11-26:
So that way also works for Dexcom? Thanks