j-kaltes/Juggluco #237: Notification appears smaller than it used to in current build

x32445345, 2024-12-11:
Hi,

First of all, thank you so much for creating this app!

I upgraded to the latest version (9.0.12) today. The last time I installed an update for the app was many months ago.
After installing the update, I noticed that the notification appears smaller than it was in the build I used previously (I don't know what exact build I used before this, unfortunately).

I looked through my phone and found an old screenshot I took in May. Here's a comparison:
[image]

It's kind of hard for me to read the number with the current build, unless I get very close to my phone.
If this was an intended change, it would be great if this could be adjusted somehow.

Thanks!

j-kaltes, 2024-12-11:
The size differs from number to number, but how they are created has also changed. The Juggluco version that can be used with Android 4.4 and higher, still uses the old method. If you only use Libre2, you can use that.
https://www.juggluco.nl/Juggluco/download.html

x32445345, 2024-12-11:
If this is something only I need, I think I could manually patch https://github.com/j-kaltes/Juggluco/blob/primary/Common/src/mobileSi/java/tk/glucodata/StatusIcon.java and compile the app myself when I want to update to the latest version.

j-kaltes, 2024-12-12:
As I said, the number icons in the status bar are not consistently smaller with the newer method, so it makes little sense to turn it back. The old icons were somewhat bolder. Also with Libre 3 you can run the no3 version as a mirror and send the data to it, so you can compare them.
[image]
[image]
[image]
[image]
The first icon is the new one, the second the old.

x32445345, 2024-12-12:
In your screenshots the difference between the versions is not as significant as what I experience on my device. I'm using a Pixel 7 Pro with LineageOS 20 / Android 13.

The height is basically the same according to what I measured (measuring the exact number of pixels can be ambiguous because of anti-aliasing):
[image]

This is the difference on my device:
[image]

On my device, the new version is 22 pixels tall, while the old version was 27 pixels tall. So for me it's not just that the text is less bold, but it's also less tall.

I am not sure why the difference is bigger on my device. The readability is impacted quite a bit on my device.