j-kaltes/Juggluco #240: Sibionics Missing

postn, 2024-12-15:
Hello. I am getting my first sibionics CGM next week and wanted to activate it with Juggluco, as per other GitHub topics and changelog I should see Sibionics in the left menu but I see photo and not sibionics. 

Is photo what I need to use? Also I assume I can activate the CGM without the official app I suppose. 

Thanks for the app and happy to share screenshots if needed.

j-kaltes, 2024-12-15:
What do you want to do with the Sibionics menu item?

Do you have Chinese market Sibionics sensors?

postn, 2024-12-15:
I have ordered one from AliExpress which by the way the pics are showing probably is Chinese version (will know for sure next week). 

Is my understanding right that if it's Chinese version then I can use juggluco if it's not then I can't use juggluco.

j-kaltes, 2024-12-15:
I have made Juggluco work with some Chinese Sibionics sensor and it turned out not to work with other Sibionics sensors. I don't know how exactly these sensors are called.
[image]
[image]
[image]
[image]

postn, 2024-12-15:
Thanks for the details. By the images on AliExpress this is how the box looks like too. If I indeed get the Chinese version of the CGM can I then use Photo (as I don't see sibionics in the menu) to scan the qr code?

j-kaltes, 2024-12-15:
You scan the data matrix by making a photo.

postn, 2024-12-25:
Just to confirm, the app worked well. Thanks again for the amazing app.

closed 2024-12-25 by postn