j-kaltes/Juggluco #242: Differences in values on Hematonix and Juggluco

KytGluc, 2024-12-17:
Hello! I am using the Sibionix sensor. At first I activated the sensor with the original Hematonix app and received data there for 24 hours. Then forcibly stopped Hematonix and connected the sensor to the Juggluco app. All the data from the past 24 hours was transferred to Juggluco, but the values received on Hematonix and transferred to Juggluco are different (meaning the values at the same point in time). How can this be?

[image]
[cgms_jug.xlsx](https://github.com/user-attachments/files/18163128/cgms_jug.xlsx)

j-kaltes, 2024-12-17:
How can this? Juggluco is said not to work with Hematonix sensors.

KytGluc, 2024-12-17:
The sensor itself is called Sibionics, but it is only activated by the Hematonix app (not the original Sibionics app) or Juggluco. And it works for 24 days on Juggluco.

j-kaltes, 2024-12-17:
If you use a Chinese Sibionics sensor for which the Juggluco support for Sibionics sensors was made, Juggluco gives the right values and the Hematonix app the wrong. Try it with the original Chinese Sibionics app. You need a Chinese telephone number for that.

KytGluc, 2024-12-17:
Thanks for your answer!