j-kaltes/Juggluco #243: Clock Development with JuggLuco ?

Hypogluco, 2024-12-18:
Thank you for your application, it has been working great for several months now.

I have created several Glycemic clocks with Libreview, Nightscout. I am developing a clock with JuggLuco in local connection with the phone and everything works, the sugar level, the trend arrow and the graphic history are displayed on my screen. However, I wanted to create a code that allows you to do without the second phone that serves as a local server, that is to say that the ESP32 card can connect directly to the Master phone and for that I need to know how the Juggluco application communicates between 2 devices? Thank you

j-kaltes, 2024-12-18:
Can't you just connect to the web server in Juggluco? 
https://www.juggluco.nl/Jugglucohelp/Nightscouthelp.html
https://www.juggluco.nl/Juggluco/webserver.html

Hypogluco, 2024-12-18:
> Can't you just connect to the web server in Juggluco? https://www.juggluco.nl/Jugglucohelp/Nightscouthelp.html https://www.juggluco.nl/Juggluco/webserver.html

Thanks for the answer,
sorry for my bad English.

I would like to display the sugar level on my clock even when the master phone is not at home. The only way I found for this solution is to have a second android device that communicates on port 8795 with the master phone. I would like to have the possibility to connect to the master phone without going through the second device.

j-kaltes, 2024-12-18:
So, you have a phone connected to the sensor that has a mobile data connection to the internet? Then you need to run Juggluco server on that clock and connect it to a modem. The modem you configure so that a port is forwarded to Juggluco server running on that clock and you connect Juggluco on the phone to the IP of the modem and the port that is forwarded to the clock.

Hypogluco, 2024-12-18:
> So, you have a phone connected to the sensor that has a mobile data connection to the internet? Then you need to run Juggluco server on that clock and connect it to a modem. The modem you configure [...]

Yes that's exactly it!!! Do you have any information on launching the Juggluco server on a raspberry pi or other ESP32 device...? Thanks

j-kaltes, 2024-12-18:
A general description is here: https://www.juggluco.nl/Juggluco/cmdline/index.html
You can compile it from this github, or download only the files needed for Juggluco server:  https://www.juggluco.nl/Juggluco/cmdline/download
You need to compile it with g++ or clang++ and it probably need libpthread for multi threading.

Hypogluco, 2024-12-18:
> Une description générale est ici : https://www.juggluco.nl/Juggluco/cmdline/index.html Vous pouvez le compiler à partir de ce github, ou télécharger uniquement les fichiers nécessaires au serveur Juggluco : https://www.juggluco.nl/Juggluco/cmdline/download Vous devez le compiler avec g++ ou clang++ et il a probablement besoin de libpthread pour le multithreading. Il n'a pas besoin de libssl, s'il se connecte uniquement à Juggluco. Il utilise c++20.

Thanks