j-kaltes/Juggluco #244: Is there support (or plans for) for the new Abbott Lingo CGM?

champeauxr, 2024-12-18:
From what I've read, the over-the-counter Lingo is built on the Freestyle Libre 3 platform, so I would expect that it could one day be supported by Juggluco, if it is not already. It was released in the US in September and is apparently available in a handful of European countries.

Unfortunately, it is not supported by the Libre app and the Lingo app is currently only available on iOS.

j-kaltes, 2024-12-18:
It looks like Libre 2, not libre 3 (see the photos at https://www.sarahhormachea.com/2024/09/18/product-review-lingo-by-abbott and https://www.cnet.com/health/medical/lingo-cgm-review-why-i-continue-to-sweeten-on-the-idea-everyone-should-check-their-blood-sugar).
Why should they have changed anything?

gui-dos, 2024-12-18:
The Lingo behaves almost like a Libre 3: https://github.com/gui-dos/DiaBLE/blob/main/DiaBLE/Lingo.swift.

In fact, when scanned via NFC, it replies with and extended `PatchInfo` of 24 bytes:

[image]

[image]

champeauxr, 2024-12-18:
There is a Libre Rio that was approved by the FDA for over-the-counter sale and was expected to be released by the end of this year but has not been. That one will be targeted at diabetics who do not take insulin.

j-kaltes, 2024-12-19:
The Libre Rio is said to be 21mm × 2.9mm (https://delivermymeds.com/freestyle-rio-cgm); that is the size of Libre 3.