j-kaltes/Juggluco #245: Support for Eversense 365?

halprewitt, 2024-12-22:
New CGM good for 1 year.  Got mine a few weeks ago.  None of the problems of the Libre 3 CGMs and more accurate. I expect this device to quickly become very popular.  There is a IOS watch option but so far, not been able to successfully read with the Android apps (Esel or xDrip). 

Thanks, hp

j-kaltes, 2024-12-23:
https://play.google.com/store/apps/details?id=com.senseonics.eversense365.us ?

halprewitt, 2024-12-23:
I was referring to your app connecting and supporting their sensor.  Their app is currently the only option available.

j-kaltes, 2024-12-23:
https://www.eversensecgm.com: _Eversense 365 gives you real-time glucose readings through a small sensor that's placed just under the skin of your upper arm by a healthcare professional._ It last for a year.
You have to find a user programmer to make it work with alternative apps.

halprewitt, 2024-12-23:
Juggluco partly supports the Libre series and just added Dexcom.  Why not add and read the Eversense 365 like Juggluco does with the Libres? Many thousands of the 365 have been installed in the past few months.   Counting the prior E model, tens of thousands are installed in US and the EU areas. I don't know about other locations.

Juggluco supports the use of more than one sensor at the same time.  This is great. We can have simultaneous data tracking. I am currently using a Libre 3 and the 365 at the sametime to compare performance.  So far, I very much like the 365 after a few weeks.

Please consider adding the 365 in to Juggluco.