j-kaltes/Juggluco #246: Always loss of Sensor after about 12 days

ThorstenWSchmidt, 2024-12-24:
First of all, I love this APP. Ist is so much better than the original one.

I use ist now on my 10th sensor. What is experience is that the sensors are dead after 12 days (give and take 4 hours).

I use Libre 3 EU version. My next sensors are Libre 3 plus, but us still have some regular ones.

With dead I mean that I cannot even find them anymore via NFC through Juggluco or another NFC Tag Scanner. It looks like the SR71SW battery is exhausted. I opened the Sensor and the battery has 0,99V. For silver oxide this equals 5%.

In the left menu -> Sensor I see status 133.
In the main menu I see state 7 (sensor not operational anymore).

Is it possible that the original software puts the sensor sometimes in a sleep mode to save energy?  E.g. during night? I would find that frightening.

Or is this a setting in my phone that is causing a higher energy consumption in the sensor? I have a Samsung Galaxy A55.

j-kaltes, 2024-12-24:
Not being able to scan via NFC, has absolutely nothing to do with the battery. I have very old used Libre 3 sensors that can still be scanned via NFC. From Libre 2 sensors you can easily remove the battery and then you can still scan the sensor (I didn’t manage to do that with a Libre 3 sensor, without demaging the whole sensor).
Also  “_stopped working. state=7_” error message has nothing to do with the battery. The Patchstatus information is received from the sensor via Bluetooth. For this the battery is needed.

halprewitt, 2024-12-24:
Amazing dates.  In the US, I don't receive any from a pharmacy longer than 3 months.  The ones Abbott sends as replacements usually are about 4- 5 months.  The accuracy issue shows up within the last 4 weeks of expiration.