j-kaltes/Juggluco #248: error when connecting via Bluetooth with status = 255

AlexOchkin, 2024-12-29:
Hi, I encountered an error when connecting via Bluetooth on the 9th day, status 255, what does this status mean? Data is received via NFC. version Juggluco-9.0.19-arm64.apk

j-kaltes, 2024-12-29:
I myself never experience a status=255 connection error. Others reported that that connection error resolved itself after some time. The only thing having to do with 255 (=0xFF) I can find is,
```c
#define  GATT_OUT_OF_RANGE                   0xFF /* Attribute value out of range */
```
[gatt_api.h](https://android.googlesource.com/platform/system/bt/+/android-7.1.1_r44/stack/include/gatt_api.h#70)

But I don’t know if that has to do with a connection error.

AlexOchkin, 2024-12-29:
Thank you for the information. A day later the sensor began transmitting readings. Now the connection is stable. Good luck.

dassystem, 2025-01-02:
I actually have that error quite often (roughly 1 out of 3 sensors). In those cases I switch to the official libre app and check what that is saying: "signal loss" / "status 57" in the protocol. Sometimes the connection comes back up after 1 to 4 hours, but sometimes it doesn't. For my next sensors I turned off "Turbo" and "Android" modes in Juggluco, because I reread through the descriptions and think this is now right choice for me. I hope that it might help.

What's really weird about this situation is, that it's so unpredictable. I always have my phone with me in my pocket, have it next to me while sleeping or showering. Sometimes sensors last the full 14 days, sometimes they drop out with "signal loss" (status=255). Sometimes after 1 day, sometimes 3, sometimes 10 days. Until now it seems to be random for me.

Electrik-rich546456, 2025-03-08:
I get this error very often im using fsl2 14 day.
I think it may have something to do with quickly changing blood sugars.
or its something to do with other BT devices saturating the frequency band and it fails to connect.