j-kaltes/Juggluco #249: Complication for any Watchfaces (just like in xdrip+)

miditkl, 2024-12-29:
Hello,

I'm currently using Jugggluco, which works perfectly. I don't like the watch faces and I want to use any watch face. On the one hand, I want it to be more personalized and on the
 other, more discreet.

I use xDrip+ to see my BS, for which I have to run 2 apps and am therefore dependent on my phone just to have the complication:
[image]
_Bottom right (Complication 4)_

If Juggluco offered a universal complication, other third-party apps would be unnecessary. I wouldn't need xDrip+ on my phone or on my watch.

The function would really be one of the best for the watch.

I am very grateful for the effort and time; the app made it easier for me to do something with the values.
Greetings

j-kaltes, 2024-12-29:
Juggluco contains three complications: an arrow, a glucose value  and an  arrow plus glucose value complication.
For that you need a small image complication slot. You can't put that in a complication slot only for short text.

[image]

miditkl, 2024-12-29:
So there is no way to implement that?

j-kaltes, 2024-12-29:
I have implemented it, but it doesn't show it well so I didn't put it into the end version of the app.

miditkl, 2024-12-29:
How did it look like? Maybe it's enough for the user 👀

j-kaltes, 2024-12-29:
Put it as the place of these weather -1 and put the -1 at the border of your watch face.

miditkl, 2024-12-29:
> EDIT: However, since xDrip+ does not have a complication for complication 5 (where I get my weather displayed), a complication for that would also be a game changer, as the values ​​can be displayed a little more clearly.

I can't change xDrip there, but I have the battery level displayed, as it is similar.
[image]
But the size looks perfect to me.

miditkl, 2024-12-29:
Alright then, I'll just make my own WatchFace, similar to the WatchFace, I am currently using, and replace the one component to an  image one.

miditkl, 2024-12-30:
It would be still nice, if Juggluco offers a short text type.

heygar, 2025-03-14:
Not really a solution, but I use the GlucoDataHandler app. It has lots of quite useful complications and it is easy to get the info from the Juggluco app. The data around the parameter is nice though. I have Juggluco, Xdrip and GlucDataHandler running all together. It would be nice to get rid of one or two of them :)