j-kaltes/Juggluco #251: Do not disturb (possibly unexpected behaviour)

postn, 2024-12-30:
Hello,

I read through and played around with the ringtone settings for alarms however I am unable to crack it (might be 100% me). 

Here's my use case: Do not ring any alarms (especially, Value available and loss of signal) if DND is on. 

What I have tried: 
1. Given the app DND permission
2. Tested the ringtone when DND is off
3. Checked the 'Disturb' option to see if the alarm rings (it does)
4. Unchecked the 'Disturb' option to see if the alarm does not ring during DND and it still does
5. Ensure 'Alarm is alarm' is unchecked

I might be doing something trivial wrong and would love to get this working. At work during meeting I normally switch on DND and do not want any alarms to ring during that time (and while asleep too). 

I am on the latest Juggluco app and on Android 15. 

Thanks.

j-kaltes, 2024-12-30:
"Alarm is alarm" and "Disturb" should be off and in Android settings you should ensure that Alarms can't interrupt it.

postn, 2024-12-30:
Thanks. This version seems to work well. 

I don't use Juggluco for high/low alarms and use it for missed reading hence the use case. Thanks for fixing the issue.

postn, 2025-01-04:
Sorry to reopen this, just wanted to inform you that the issue is back with the latest google play store update - 9.0.22-phone-li3-si-x-g (also vibrate would still 'vibrate' in DND mode).

j-kaltes, 2025-01-04:
Nothing is back. You are now talking about **_"vibrate"_**, earlier you were talking about sound. Nothing has changed with vibrate, it was always audible in Do-Not-Disturb mode. In the last version, sound can be set not to disrupt Do-Not-Disturb by turning off "Alarm is alarm".

postn, 2025-01-04:
Probably I wasn't clear. The notifications still ring even with alarm is alarm being unchecked and disturb unchecked. Also what I meant was the Google drive apk you provided worked (no sound during DND but vibrate was still on).

Anyway I just wanted to post and update. No action needed. Thanks.

closed 2025-01-04 by postn

j-kaltes, 2025-01-04:
Notifications? Notifications are a matter of the Android settings, not of my app.

postn, 2025-01-04:
Sorry I meant sound not notification. I will probably clean cache and try again. No worries. Thanks.