j-kaltes/Juggluco #252: Can I change phone "mid sensor" (i.e. without losing the time left on my sensor)?

jaimet, 2024-12-30:
I am currently successfully using juggluco on "phone1" with a european libre 2 sensor. I want to stop using phone1 and start using "phone2" but I do not want lose the 10 days left on my current sensor.

If I start scanning the already-initialised sensor using phone2:

a) Will NFC scans on phone2 work?
b) Will phone2 be able to make a bluetooth low energy connection with the sensor?

TIA, Jaime

j-kaltes, 2024-12-30:
This will give no problem at all if you turn off left menu->sensor->"_Use Bluetooth_" on the old phone. You can also transfer the data from the old phone to the new phone: https://www.juggluco.nl/Juggluco/index.html#newphone

jaimet, 2024-12-30:
> This will give no problem at all if you turn off left menu->sensor->"_Use Bluetooth_" on the old phone. You can also transfer the data from the old phone to the new phone: https://www.juggluco.nl/Juggluco/index.html#newphone

Thank you! :+1: :pray:

closed 2024-12-30 by jaimet