j-kaltes/Juggluco #253: Libreview: postgetauth: status=39 reason=invalidAuthenticationRequest

Odiz2, 2024-12-31:
When trying to send data to LibreView on my Galaxy S24 I get the following error in the Juggluco/Exchange Data/LibreView Settings:

postgetauth: status=39 reason=invalidAuthenticationRequest

I can't retrieve the account ID either but got it through the patched Libre3 app.

I have the latest version of Juggluco(9.0.19)

j-kaltes, 2024-12-31:
I have never seen that error message. What have you done to create this error message?

Odiz2, 2025-01-01:
Unfortunately I can't say, after submitting this post it suddenly worked. 
I have never been able to get the LibreView ID(can't remember if the same error message appeared then) so I just used a random number. That was not a problem until the latest version of LibreView finally supported my phone and I was able to activate a Libre 3 sensor in that app. In order to read that sensor with Juggluco I needed the ID (which I managed to get by installing the patched LibreView app, get the ID and removing that app again).
Getting this far I tried to also upload the data to LibreView which didn't work previously when using the random ID.

The below are a few things I tried but subsequent test of Juggluco always gave the above error regardless. 
1) Restart after installing the latest version of Juggluco
2) Force stop LibreView
3) Changed my LibreView password from some auto-generated one(with lots of weird characters) to a more human-readable one

As mentioned none of this immediately fixed the problem - maybe Juggluco will not retry instantly, even when pressing the "Send Now" button, so the error may only disappear later when the app sends data to LibreView as part of its normal schedule.