j-kaltes/Juggluco #254: Glucose line chart colors change with every new sensor

DarthRainbows, 2025-01-06:
I set the chart line colors manually to my preferred colors. Every time I start a ne sensor, the app changes the colors used for that sensor. I'm not sure if this is a bug or intended behavior, but either way, it would be nice to not have to reset the colors every single time.
[image]

j-kaltes, 2025-01-06:
Juggluco tries to give the curves of every sensor a different color, so that you can differentiate between them. If you don't like the default colors, you have to change them every 10, 14 or 24 days depending on what sensor you use.
Giving them all the same color would be a very bad idea.
[image]

DarthRainbows, 2025-01-06:
I disagree that same color is a bad idea. If you're only managing one sensor, there's no color confusion. I can see an argument for not making this the default if the user base manages multiple sensors, but a checkbox for persisting the color choices would be feasible.

Changing colors every time is doable, but not always easy, especially for the visually impaired. If the random color chosen is hard for the user to see, it can be challenging to even identify where the line is in order to tap it to change the color.

bentonjames-lab, 2026-11-15:
is there a way a user selectable option can be checked to retain custom colors set for the graph lines, after a sensor change?  
I only monitor 1 sensor, I dont need a color seperation between sensors, the gap in data while new sensor is starting is a very good way to keep previous and new sensors visually seperate.

I understand the need.some would want seperate colors at sensor change, but I know would like an option to retain them permanently once I set them to my preferences