j-kaltes/Juggluco #255: Libre 3, starting a new sensor has inconsistent behavior

DarthRainbows, 2025-01-06:
Sometimes when starting a new sensor, Juggluco shows a popup asking to add the sensor end date to my calendar. Other times, there is a full-screen display that says the sensor will be ready in 60 minutes, without the option to add an event to my calendar. As far as I know, I'm not doing anything different to prompt different behavior. I'd prefer to have the workflow that sets up the calendar event every time.

j-kaltes, 2025-01-06:
Juggluco only shows a _Ready for use_ screen with the option to start the calendar app only the first time you scan a Libre 3 sensor:
[image]
With Libre 3 sensor, it will be shown again when you kill Juggluco, start it again and scan the sensor.

DarthRainbows, 2025-01-06:
I just started a new sensor last night, and I did not get that _Ready for use_ screen. Killing Juggluco, restarting, and rescanning did get it to show. Still, should have been shown the first time.

j-kaltes, 2025-01-06:
I never experienced that is wasn't shown and looking at the code, I can't see how that could be possible. But I don't use Libre 3.

surgicalcoder, 2025-01-16:
I seem to be suffering from this inconsistency as well, sometimes I get the "Sensor ready for use" popup and sometimes I don't - is it possible to get a button under Senor, Info that adds either the offical or sensor end date to calendar?

closed 2026-01-30 by j-kaltes