j-kaltes/Juggluco #256: Libre3 sensor ends early

Havanna22, 2025-01-17:
Have a Samsung S21 with Libre 3 and Juggluco. Everything works fine except that the sensor ends after 11-12 days with error 7.
Not a major issue but does add up over time when losing days for each sensor.

j-kaltes, 2025-01-18:
That is something of the sensor and has nothing to with the connection of Juggluco with the sensor. Contact Abbott.

Havanna22, 2025-01-18:
I don't think so since it's consistent since I changed to Juggluco.
I have had Libre3 for years and only had 3-4 sensors failing in that time.
With Juggluco the last three sensors have stopped early.

maheini, 2025-01-19:
I recently had 2 libre3 sensors in a row with similar issues, one of them was already sent from Abbot as a replacement sensor. Don't know if the cause is triggered by a hardware (libre) or software (juggluco) issue yet. 

Android 15 (Pixel 9)
Libre 3 EU
Juggluco 9.0.26

Symptoms
After having no issues at all, the readings will slowly drop and stay below 4mmol after some time. I can still notice some movements in the juggluco graph if my BG is rising or falling, but it seems to be weak and low. (always below 4mmol, 80mg/dl). Switching back to libre 3 app (using this faulty sensor) didn't solve the problem.

First one failed after ~11 days
Second one failed after 3.5 days

I personally suspect a Libre 3 issue or problems with android15. I will keep having a eye on this issue.

j-kaltes, 2025-01-19:
You need to differentiate sharply between connection errors on the one side and quality of measurement, sensor errors and replace sensor errors at the other side. The connection errors can be influenced by the app, the quality of the watch and phone and the Android version, but other errors not. The other errors can also be reproduced by Abbott’s Libre 3 app running on a different phone. When you scan the sensor after you have turned off WIFI you can retrieve all history values from the start of the sensor out of the memory of the sensor in Abbott’s Libre 3 app running on a different phone.
The sensor runs and measures independently of the app communicating with it.

maheini, 2025-01-19:
Hi @j-kaltes 
Thank you. I'm aware of that, just wanted to contribute my recent experience in order to help gathering informations. I god a new batch of sensors recently and will start the first one of those tomorrow - we'll see then if the sensor keep ending with bad readings at the end -> and therefore ends early (after some time the sensor will detects it's invalid readings and finally stop).

maheini, 2025-02-03:
Turns out my latest sensor (new order) is working perfectly fine again, just wanted you to know. Didn't changed anything (on purpose).

This was most definitely a sensor issue, I guess the same counts for @Havanna22